prusa3d/Prusa-Firmware-Buddy #4343: [ENHANCEMENT] Increase I2C M260 buffer size

me25542, 2024-12-14:
### Printer model

All that support I2C

### Describe the enhancement

The current implementation of the M260 command is great, and allows users to communicate with external devices. However, the maximum buffer size of 32 bytes can be limiting. I don't know if this is a hardware restriction, or in software. I did look in the `STM32F427ZIT6`'s datasheet, and couldn't find anything. If this is a hardware limitation, let me know and close the issue.

### Expected functionality

The print file would be able to add, say, up to 256 bytes to the buffer, and send it all in one command.

bkerler, 2025-01-09:
As we are discussing internally if sending 256 bytes in one round does make any sense instead of sending 8*32 byte packets in sequence, can you please let us know your specific use case ?

me25542, 2025-01-09:
I have a custom "smart" enclosure connected to the printer via I2C, and for most commands from the printer to the enclosure 32 bytes is enough. But, for setting a print name to be displayed on the enclosure, 32 bytes is often too short. I have implemented a workaround for this, enabling the data to be split into multiple 32-byte packets. However, it would be easier to send a single larger packet. This is low priority since implementing the workaround on my enclosure, but it would be nice. ☺️

bkerler, 2025-01-10:
Thanks. PR is already written, let's see how the dev gods will decide :D

me25542, 2025-01-11:
ok lol