prusa3d/Prusa-Firmware-Buddy #4344: [BUG] Octoprint related issues

tkircher, 2024-12-14:
### Printer model

MK4

### Firmware version

6.1.3

### Upgrades and modifications

None

### Printing from...

Octoprint/USB

### Describe the bug

After starting a print from octoprint, I noticed that I had the wrong filament loaded. It had just probed the print bed and the temperature was set to 170. I pushed the pause button on the Prusa screen, then pushed the settings button, went to change filament, and successfully unloaded the filament. Then I went to load the filament and noticed the temperature was still at 170, and the extruder was clicking of course because the hotend wasn't hot enough. At that point I just hit reset, but it seems like it should have asked what kind of filament I wanted to load first and then set the temperature correctly. This is probably not a printer state that testing is likely to have found.

danopernis, 2025-02-13:
@tkircher None of us are actively using/testing Octoprint so bugs like these can probably happen. Commenting to prevent stalebot from closing and assigning "good first issue" label in hopes that somebody from the community provides a fix.

tkircher, 2025-02-13:
> [@tkircher](https://github.com/tkircher) None of us are actively using/testing Octoprint so bugs like these can probably happen. Commenting to prevent stalebot from closing and assigning "good first issue" label in hopes that somebody from the community provides a fix.

Absolutely nobody in the real world is using PrusaLink or PrusaConnect. It's fine if your team isn't using Octoprint but all of your customers are.

danopernis, 2025-02-13:
Look, as soon as you start instructing printer using serial line, all the bets are off. You press "pause" while serial printing is in flight. At that point printer is confused about what it is supposed to be doing. Should it control the temperature based on the octoprint instructions on serial line? Or should it control it based on user interactions from the menu? It is not an easy task to decide. It takes time to investigate and make sure you don't break some other use-case. Arguably, this time could be spent on more pressing issues.

I'd be interested to see actual number of users myself. If it turns out that, as you put it, "all of your customers are (using Octoprint)" then we should definitely prioritize that.

tkircher, 2025-02-13:
That's not a very good example - in that case when you hit the pause button the printer should send octoprint a @pause command because octoprint is in charge, all interface commands on the printer itself should pass through to octoprint and only octoprint serial commands should be interpreted. If you cancel the print and octoprint releases control then it goes back to direct interface control. This has been figured out absolutely ages ago and only Prusa does not do it correctly.

danopernis, 2025-02-13:
So you basically want us to enable Marlin's `HOST_ACTION_COMMANDS` feature? AFAIK that was never enabled in the 32-bit firmware, not sure about 8-bit firmware. It may not be so easy to enable now, as the firmware at this point quite diverged from the original Marlin source code. We have reimplemented a lot of the G-Codes (and other stuff from Marlin). Not saying it is not doable, and I am still willing to merge community pull requests, should anybody want to start hacking on this. At least now it is more clear what it is that you are wanting us to do.

Again, comments like "only Prusa does not do it correctly" seem factually wrong, overly aggressive and add nothing useful to the debate. I am quite certain there are other 3D printers not supporting those. Personally, this makes me not want to work on such features even more.

closed 2025-04-23 by github-actions[bot]