prusa3d/Prusa-Firmware-Buddy #4350: [BUG] Flexible filament sometimes fails to home using the accelerometers

hackwin, 2024-12-16:
### Printer model

MK4

### Firmware version

6.1.2

### Printing from...

PrusaLink

### Describe the bug

Flexible filament is loaded into the printer.  The printer attempts to home each axis before it can print.  The printer mistakes the stretching/knocking of the spool as a signal that it has reached the end of the axis.  It can be fixed by unspooling more flexible filament so that the printer reaches the end of the axis.

### How to reproduce

Filament tested is Microcenter Inland TPU 95-a hardness

### Expected behavior

If flexible filament is loaded, maybe there is some type of movement that can unspool more during homing.  Or, allow homing to be less sensitive to small acceleration signals of knocking flexible filament.

MichaelJLew, 2024-12-19:
Sometimes happens to me too (MK4). A solution is to have a loose section of filament between the spool and the extruder, but a firmware fix would also be nice.

CZDanol, 2024-12-20:
Internal ticket: BFW-6491