prusa3d/Prusa-Firmware-Buddy #4353: [BUG] Prusa XL instability.

rvalle, 2024-12-19:
### Printer model

XL 2T

### Firmware version

6.1.3

### Printing from...

USB and Wifi / Prusalink

### Describe the bug

My XL is unstable with head crashed and filament sensor issues.
The printer has been taken in by Prusa and has been verified to be mechanically and electrically perfect.
Yet the printer remains to fail.
For a detailed log of the case check this [forum post](https://forum.prusa3d.com/forum/original-prusa-xl-tool-changer-hardware-firmware-and-software-help/failing-xl-what-do-you-think-is-the-problem/#post-730177): 
There are many other report by users and bugs on this repo consistent with my experience: 

#4235
#4329 
#4304 
#4259 

While these issues tent to be interpreted as mechanical, in my case the issues persists after Prusa support has completely verified the printer. They are NOT mechanical. 

### How to reproduce

Normal usage of Prusa XL printer.

### Expected behavior

Head should not crash during print or calibration
Layers should not massively shift
Filament detectors should work
Printer should not auto-reset

CZDanol, 2024-12-19:
Hello, please start by updating to 6.2.0-alpha2 firmware.

SaschaKP, 2024-12-25:
anyway, this is something I noticed, to remain in topic:

If you end a print and start another one without rebooting the printer, the homing runs slower, after the second print, if you start another print (so this is the third one) this problems became true and the homing takes ages, so just reboot the printer, and let's hope they fix it, since this is something I see from some time.

CZDanol, 2025-01-02:
@SaschaKP slower in what sense? Are some specific moves slower? Or does it do more retries?

SaschaKP, 2025-01-09:
> @SaschaKP slower in what sense? Are some specific moves slower? Or does it do more retries?

Hello!
Exactly as I said, the homing is slower, about 2 times slower, and the third time a new print starts without any reboot it gests slower (4 time or more) and the head begins to crash and print is not done in the correct way, it's easily reproducible, start a print, let it finish (even a print of a couple of seconds), start the same print again, this time homing will be slower and slower, every new print gets slower.

SaschaKP, 2025-01-09:
Just to add more info, but maybe not related to what I wrote above, the following consideration:

Does the hardware allow the use of an UPS? I mean: not fully sine-wave, but emulated sine-wave?
If this is isolated or maybe even not isolated, but only sparse, maybe this could be the reason why some users have those problems?
I will do some more tries and give more infos as my new print starts, considering that I have in fact no reason for the additional UPS, since I have a solar+inverter+backup battery for the whole place, this is something I was already considering.
I will update (but not so soon, I have a print going on, maybe tomorrow)

CZDanol, 2025-01-09:
Hmm, perhaps some gcode injected in your file that reduces print speed? Can you check your profiles?

The homing should be unaffected by this though, that is worth investigating.

As for UPS support: https://help.prusa3d.com/article/ups-backup-for-your-printer_266023

CZDanol, 2025-01-09:
@SaschaKP Would you please provide some of the gcodes that you printed? It would possibly make it easier for us to reproduce the bug.

SaschaKP, 2025-01-09:
No problem at all, with the same settings, this is something I printed, but exported in GCODE for readability:
[example_GCODE_FLEX_XLIS_3h55m.zip](https://github.com/user-attachments/files/18367689/example_GCODE_FLEX_XLIS_3h55m.zip)

SaschaKP, 2025-01-09:
> Hmm, perhaps some gcode injected in your file that reduces print speed? Can you check your profiles?
> 
sorry for the double post @CZDanol 
The profiles are original, but again, maybe even If I resetted anything, something couldn't be good, so I posted the GCODE just to be sure everything is OK.

CZDanol, 2025-01-09:
Actually, I've just realized that this might have been linked to a bug that we've fixed in 6.2.0-RC1 that we've released just tuday: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-RC1 . Might you check it again please on this version?

SaschaKP, 2025-01-09:
> Actually, I've just realized that this might have been linked to a bug that we've fixed in 6.2.0-RC1 that we've released just tuday: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-RC1 . Might you check it again please on this version?

I will let you know ASAP, thanks

SaschaKP, 2025-01-10:
Solved by https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-RC1

CZDanol, 2025-01-10:
Okay, thanks for the info @SaschaKP . I will keep the thread open in case we get some new information - in retrospective, it would probably have been better if we created a separate issue for your problem.

closed 2025-03-19 by github-actions[bot]