prusa3d/Prusa-Firmware-Buddy #4356: [ENHANCEMENT] Add Minimum Z Parking Height for Filament Out Changing

Xstell132, 2024-12-21:
### Printer model

MK4S

### Describe the enhancement

With the current fan setup on the MK4S it's incredible difficult to clear purged filament when changing out spools when the Z axis is close to the bed (say on short prints or early in a print job). You need to be extra careful not to burn your hand on the nozzle or get your hand pinched when the extruder does it's last quick little purge before the hotend travels back to resume printing.

### Expected functionality

- When filament out is detected, set the Z parking height to a minimum height of 100mm to allow enough room for the user to safely clear excess filament during the spool changeout & purge process.

- If at or above 100mm, run the changeout program as currently done as there should already be sufficient room.

CZDanol, 2024-12-21:
Hey, I believe this should already be in place - either please check 6.2.0 or maybe it will be in the next version, not sure.

Xstell132, 2024-12-21:
I looked through the change logs and issues. I saw a couple topics mentioning the parking position in regards to left, center, right positions but nothing talking about setting a minimum Z height yet. Where would this be documented?

CZDanol, 2025-01-02:
Well, if you wanted to find the source of it, it would be somewhere in the commits - but I would just recommend flashing the 6.2 and trying it out.

closed 2025-03-11 by github-actions[bot]