prusa3d/Prusa-Firmware-Buddy #4363: [BUG] Prusa Link Raspberry Pi W 2 No lan access

alexkots, 2024-12-24:
### Printer model

MK3S+

### Firmware version

3.14.0-8066

### Upgrades and modifications

Prusa Link

### Printing from...

Prusa Link

### Describe the bug

Powering up the printer and after Prusa Link boot, I had a message **No lan access**. I switched off and then on the printer, but after new boot I had the same message. A Scan of my LAN with Advanced IP Scanner, verified that Prusa Link had not been connected. As I was searching the internet for a solution, sometime I saw the message **Prusa Link OK**.  Trying to connect with the hostname I have ginven to it at setup, There was not any connection. And the IP that I have reserved in my rooter for the printer MAC was not the same as before. Checking again the LAN, I found that printer has changed MAC! I configured my router with the new MAC, but still unable to log with the "server" name. Somewhere I found that I can add a file to the SD card, where I can write the SSD and Password for the Wifi plus server name. I did it and the printer "returned" to original status (and original MAC). But now it shows me a warning **HTML error 5xx** (for connect.prusa3d.com). Do you have an idea what could be wrong?

### How to reproduce

No, I can't reproduce it, I did not found any sequence of actions that is generating this error.

sl1pkn07, 2024-12-27:
if use NetworkManager inside:

https://gist.github.com/ctubbsii/a3f15f7806830945a0b6c68ec0e23afa

if not, search in internet what distro use prusalink and how stop randomize MAC address in that distro (maybe is Debian)

greetings

alexkots, 2024-12-28:
My PC has Windows 10, not Linux (and I am not familiar with that). So I think I can't do anything from what you sugest. Is there any file in the SD card (that can be edited by Win 10) where I can declare something that will fix MAC address?
I do not understand why Prusa Link is been set with a double(?) MAC address (at the moment I have seen two MAC addresses). Typically any device has to have **one** MAC address and not switching between two. It is very anoying ...

sl1pkn07, 2024-12-28:
reading better the fist post. this is not the place for this. you need open the issue in the prusalink repository. because the MK3S+ not use the buddy firmware, and prusalink is installed in external device

greetings