prusa3d/Prusa-Firmware-Buddy #4365: [BUG] Screen flickers / goes black

thirtythirtyal, 2024-12-26:
### Printer model

MK4S Kit

### Firmware version

6.2.0-alpha3-8702

### Upgrades and modifications

None

### Printing from...

USB Drive

### Describe the bug

I sent an email to Prusa again about this.  Here is the content of that email.

Still having an issue with screen black-out / flicker when I turn the printer on.  I've read the forums and tried everything I saw there (ribbon cable, connectors, firmware etc) and also tried what was recommended by your staff in a chat recently.  The ribbon cable is plugged in correctly at both ends.  No damage or bent pins. The ribbon cable is flat and not folded sharply anywhere.  I have the latest firmware.  I'm still dealing with this annoying issue.  When I turn the printer on, the screen shows loading, the bar gets about halfway across the screen, then it goes black, flickers various distorted patterns then goes black and the fans turn off.  If I turn the printer off and back on five or six times it eventually stops doing that and works fine.  It's been on for several days now with no printing issues.  When I turn it off, I'll have the screen flicker issue again.  It's really annoying and I'm worried it won't work at all eventually. Any new suggestions?

### How to reproduce

yes, by cycling the power.

LarGriff, 2024-12-26:
Similar to another issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3947
Some people have had success by adding snubber resistors.  Others by grounding the MMU board to xBuddy PE.

CZDanol, 2025-01-02:
Hello, there is no alpha3 release of 6.2.0 - I assume you meant alpha2.

There should be a new release candidate out soon (within a few weeks) with an option to reduce display refresh speed - that might resolve the issue (it is not a fix, just a check for us to determine what is the cause).

For the meantime, you can try re-seating the display connector on both sides. If that doesn't help, you can also try contacting our support.

Prusa-Support, 2025-01-31:
Thanks for bearing with us.
We haven't seen more comments on this issue so the problem was probably "solved" by FW 6.2.0-RC1 or via rewiring.

The final release [6.2.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0) is officially available and _reducing the refresh rate should bring significant improvement_ in similar cases.
Regardless of this, it seems that we can close this issue as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4283.

Michele Moramarco
Prusa Research

closed 2025-01-31 by Prusa-Support

thirtythirtyal, 2025-01-31:
I've had no issues since I went through and unconnected/reconnected the
three ends of the yellow wire.  I like the MK4S a lot.  I plan to buy the
Core 1 as well.

Thankyou for the follow up
Al