prusa3d/Prusa-Firmware-Buddy #4368: [BUG] USB drive or file error

BixlerVDL, 2024-12-26:
### Printer model

Prusa XL

### Firmware version

6.1.3

### Printing from...

Prusa connect

### Describe the bug

Occasionally our Prusa XL printers get the USB file drive error.
Once a printer is in this state, you have to press the knob and then you can press reprint to start the print that was intended to be printed.
To get the printer out of the error state, the error can only be acknowledged on the printer, not remotely. 
It would be best if the error does not appear in the first place, but if it does it would be good if the error can be acknowledged remotely. Because now I can't print if I am away and the printers are giving the error.

We are using the USB sticks that came with the printer, there are only 8 BGCODE files on the USB stick.

### How to reproduce

Occasionally our Prusa XL printers get the USB file drive error.
Once a printer is in this state, you have to press the knob and then you can press reprint to start the print that was intended to be printed.
To get the printer out of the error state, the error can only be acknowledged on the printer, not remotely.

CZDanol, 2025-01-02:
Hello, this should be fixed in 6.2.0

CZDanol, 2025-01-04:
@FSXLiveriesdotcom I see, please describe your problem a bit more then. So:

1. You are still occasionally getting an USB error?
2. Are you able to resume every time?
3. Do you have to do anything to resume, or is it enough to just press the button?
4. Can the resume by done through the Connect, or do you have to do it manually?

CZDanol, 2025-01-06:
@FSXLiveriesdotcom thanks, so on 6.2.0-alpha2, this only happened to you once so far, right? And pressing the button didn't help, you had to reinsert the drive?

closed 2025-05-19 by github-actions[bot]