prusa3d/Prusa-Firmware-Buddy #4369: [FEATURE REQUEST] Allow for user to manually define loaded filament types

p27182, 2024-12-26:
### Printer model

XL

### Describe the feature

After loading a new firmware none of my printer's settings were saved for reasons unknown to me and I learned that I have to click through the menu to unload then reload the filament that already loaded five times just so the printer is aware of what type of filament is loaded.

### Expected functionality

Just let me tell the printer what filament is loaded...

danopernis, 2024-12-30:
This doesn't seem like a feature request, rather it would be bug. What firmware versions are we talking about?

p27182, 2024-12-30:
> This doesn't seem like a feature request, rather it would be bug. 

i thought of it more as a state of the machine that was not considered during ui control layout but it's entirely debatable...  id be happy to edit it if means it might get some attention 👌

> What firmware versions are we talking about?

this most recently was on the 6.2alpha2 fw but im pretty sure its missing from previous versions too

danopernis, 2024-12-31:
I vaguely recall a specific bug in 6.2.0-alpha related to missing EEPROM migration related to upgrades from older firmware versions, so this might actually be already fixed. Please wait for the 6.2.0-beta and see if it fixes the problem. The correct procedure to let the printer know about the filament type is to perform the filament load. The printer should remember this between printer updates; if it doesn't, it's a bug.

CZDanol, 2024-12-31:
There was BFW-6236 that implemented loaded filament migration  because of potential backwards incompability of the config store records, but that one has been merged after a2 was released. Prior to that, I don't know of any loaded filament changes.

Even that migration should not result in clearing the loaded filaments unless there has been a FW downgrade somewhere.

CZDanol, 2024-12-31:
@danopernis what the guy is referring to is that the config store has been COMPLETELY reset, not just the loaded filaments. 
So there's two points actually:

- Report that upgrading a FW clears the config store (I've heard someone mentioning that on the reddit, too)
- Feature request for when this happens (or for other situations) to add a way to set a currently loaded filament without having to go through the loading procedure.

CZDanol, 2024-12-31:
Created BFW-6503 for the eeprom wipe bug and 6504 for the set loaded filament feature request.

closed 2025-04-29 by github-actions[bot]

p27182, 2025-05-09:
> I've just realized that we now have the `M865` gcode that allows you to override the currently loaded filament. You can for example to `M865 F"ASA" L1` to set the Tool 2 to ASA

based on my attempt:
```
Send: M865 F"ASA" L1
Recv: echo:Unknown command: "M865 F"ASA" L1"
```
can you advise?