prusa3d/Prusa-Firmware-Buddy #4370: [FEATURE REQUEST] Use USB-C port for flash drives

me25542, 2024-12-27:
### Printer model

xBuddy based; MK4(s), MK3.9(s), MK3.5(s)

### Describe the feature

Using the USB-C port on the xBuddy board for USb flash drives would allow mainly for the USB drive to not be exposed (unsightly, and vulnerable to being hit accidently, broken, etc.) when the primary means of printing is remote via PrusaConnect or PrusaLink. I don't know for certain if this is possible with the hardware on the xBuddy board (to use the port in `Host` mode), but after looking through the schematics I think it *might*.

### Expected functionality

The printer would be able to use USB flash drives inserted into the USB-C port for printing, etc.

neofir3, 2024-12-30:
Yes, that would be great if that were possible. Then the data wouldn't always have to go through the xLCD cable to the front to the USB port. This might also make the (B)GCode upload faster. At the time, I actually assumed that you could also plug a USB stick into the back. But unfortunately, so far that's only been possible in front of the display. Although an xBuddy board with an NVMe slot would also be a dream. :)

danopernis, 2024-12-30:
@me25542 not high on the priority list, but looking briefly at the schematics, it might be possible. Looks like a nice hacking project for anybody interested :slightly_smiling_face: 

@neofir3 Buddy, XBuddy and XLBuddy boards are already shipped to thousands of customers so they are not likely to contain any new hardware. I can't disclose details about next generation of the motherboard.

Tupson444, 2024-12-31:
Good idea. If it is possible, It can be used as a backup if the front port breaks (I don't know how likely that is to happen, but anyway), or if you only have USB-C type flash at hand.

If it is not limited by hardware, then simply why not :)

me25542, 2025-05-05:
Still want this issue open.

me25542, 2025-09-08:
Still would like this.

me25542, 2025-11-10:
This issues appears to be stale; if anyone else still wants it open please comment, otherwise I will close it.

me25542, 2026-09-28:
This issue should stay open.

suomi35, 2026-10-28:
This issue should stay open.