prusa3d/Prusa-Firmware-Buddy #4371: [FEATURE REQUEST]A Setting to set the distance between the MMU and the Nextruder (For longer PTFE Tubes)

MrPresident2, 2024-12-27:
### Printer model

MK4,MK4S, CoreONE

### Describe the feature

We were chatting about it in the forum and on Reddit. A lot of us are mounting the MMU outside of the enclosure, there are many reasons to do it, for ease of maintenance, ease of loading filament, the ability to eliminate the buffer system, if the enclosure is heated and you're trying to keep things from overheating, and many more.

while trying to mount the MMU unit at any distance longer than what is specified in the firmware, the MMU will change speeds and will start loading the filament really really slow, (until the filament sensor on the Nextruder is triggered), thus making a filament change exponentially longer.

Giving us the ability, to change that distance in settings, will cause the printer to work faster, better, and will allow us to keep printing and using the printer in a heated enclosure.

### Expected functionality

The feature will be a setting you can edit at the printer's settings interface. The user will be able to measure the new length of the PTFE tube, from the MMU unit to the Nextruder, I'm MM, and will put that input in that setting page. Thus allowing the MMU unit to push the filament after, at the desired length, at the same speed until the Nextruder filament sensor is triggered.

Same functionality as we have right now, but with the ability to edit the length of the PTFE tube connecting the MMU and the Nextruder.

DRracer, 2025-01-02:
Please use MMU's register 34. Beware - it is persistent (stored in EEPROM) in the MMU
```
| 0x22 34 | uint16 | Bowden length  | 341-1000 | 168h 360 | unit mm | Read / Write Persistent | M707 A0x22 | M708 A0x22 Xn
```

For reference see https://github.com/prusa3d/Prusa-Firmware-MMU/blob/main/src/registers.cpp

Our help pages are missing this register, we'll have it fixed soon.

DRracer, 2025-01-04:
https://help.prusa3d.com/article/registers-mmu-mmu3_511780 has been updated

closed 2025-01-04 by DRracer