prusa3d/Prusa-Firmware-Buddy #4372: [QUESTION] Why is the GitHub master code of 6.1.0 while current firmware version is 6.1.4?

bthe0, 2024-12-27:
### Question

Why is the GitHub master code of 6.1.0 while current firmware version is 6.1.4? Thank you!

TouchMonkey, 2024-12-28:
I have no clue what logic they use when adding commits to the master branch, but if you want a specific release it should be available as a tag.

[image]

Which will take you to a page where you can search the tags. Note that "git fetch" doesn't always pull all tags. If one is on the web but not available on your system run "git fetch --tags --all"

[image]

bthe0, 2024-12-28:
> I have no clue what logic they use when adding commits to the master branch, but if you want a specific release it should be available as a tag. [...]

Thank you! I'd still like to know the logic behind master branch if possible :)

CZDanol, 2025-01-02:
Me too :D

closed 2025-03-11 by github-actions[bot]