prusa3d/Prusa-Firmware-Buddy #4373: [BUG] XY calibration fails

japato, 2024-12-28:
### Printer model

Mini+

### Firmware version

6.2.0-alpha2

### Printing from...

none

### Describe the bug

XY test fails on Mini+ on Y axis test. On previous fw (6.0.4) passed.

danopernis, 2024-12-30:
@japato does the test pass again after you rollback to previous firmware? If it does, can you identify the first firmware version which fails for you? [List of all releases](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases).

japato, 2024-12-30:
Hello,

Yes, after rollback to 6.0.4 test passed again (I hadn´t 6.1.2. at that moment). 

This test fails first with alpha2 after some works on „Mini+ Stabilizer w/ integrated carry handle“ installation.

After successful test on 6.0.4 I went back to 6.2.0-alpha2 again without update to 6.1.2.

I can try 6.1.2 after date 3.1.25

 

Jiří Tomek

japato, 2025-01-02:
> Odesláno: pondělí 30. prosince 2024 16:01
> Komu: prusa3d/Prusa-Firmware-Buddy ***@***.***>

Hello again,
so I have tryed the same procedure with rollback to 6.1.2 release.
Callibration of XY passed normaly again.
Next in czech:
kalibrace na 6.2.0-alpha2 probíhá tak, že na ose X několikrát projede tam a zpět a odfajfkuje ji jako dobrou. Oproti tomu na ose Y odveze vozík dozadu, jede vpřed, po nárazu ještě chviku vrčí motor, ale pak kalibraci ukončí s chybou.
Tož tak
zdraví
Jiří Tomek

encip, 2025-01-09:
Duplicate of #4218 

Issue persists with 6.2.0-RC1

CZDanol, 2025-01-09:
Closing as a duplicate of #4218 . Please continue your disccusion there.

closed 2025-01-09 by CZDanol