prusa3d/Prusa-Firmware-Buddy #4374: [FEATURE REQUEST] Display the Repetier ICON not the Octoprint ICON when USB control is Repetier Server

gjpc, 2024-12-29:
### Printer model

XL

### Describe the feature

Currently when I printing models on my XL 5T a smiling cephalopod is prominently display on the XL's LCD panel. I would like to see the Repetier tesseract.
[image]

### Expected functionality

Somewhere in the Prusa firmware there lurks a happy Octopus. The Repetier tesseract should be displayed if Repetier server is controlling the printer.

danopernis, 2024-12-30:
That is a correct observation. However, we can't possibly have a distinct icon for every piece of software that happens to hijack printer's serial port. Therefore, I propose an alternate solution: I vote for killing the octopus. Only question is what to display instead of it. Part of the problem is that we _need_ some screen to prevent unwanted interactions during the print, like using the Control -> Move Axis menu or filament load/unload.

As for the octopus habitat in the source code, it lives [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/src/gui/res/png/serial_printing_172x138.png). This might be useful for people building their own firmware.

Also worth noting, there is an option to disable serial printing screen for expert users (implemented in https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/4071).

gjpc, 2024-12-30:
@danopernis Thanks for the quick reply.  Alas, the experimental menu is not available unless I want to maintain my own build of the firmware, which I do not. Besides, I would just change the graphic rather than enable the experimental menu in my build. I'll come up with a non-secular graphic to dispose of the cephalopod and submit a PR.

gjpc, 2024-12-30:
https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/4377

danopernis, 2024-12-31:
@gjpc Thanks for the effort, but I am afraid I have to consult this with our UX department. I will get in touch with them after new year and hopefully resolve this.

CZDanol, 2025-01-04:
IIRC we've been having some requests to show standard printing screen during serial print, perhaps what would be the way?

gjpc, 2025-01-04:
@CZDanolI am suggesting 
[image]
In my PR #4377

gjpc, 2025-03-06:
Still waiting to see if my PR is accepted.

gjpc, 2025-05-12:
OCTOPRINT is not the only USB control program. Please generalize the display.

n8many, 2025-05-26:
Not to swoop under you @gjpc, but I worry the icon you proposed could run afoul of the USB consortium being too close to their official trademarks (and perhaps could be some of the reason for slowness to merge). I have created a pull request for an alternate that uses the  USB trident mark (which from my research seems to be unprotected) and better matches the Prusa style guidelines.

Also @danopernis, also happy to send your UX team the original vector file (Affinity Designer) if they need to make modifications or better rasterize it. The smaller icon is a simple pixel drawing, so no other source file is needed.