prusa3d/Prusa-Firmware-Buddy #4375: [BUG] Prusa Connect w/ XL - Cancel Object Fails Error 422 w/ large number of objects

oriddlero, 2024-12-30:
### Printer model

XL

### Firmware version

6.1.3+7898

### Printing from...

Prusa Connect

### Describe the bug

Cancel object is a critical function to me, I am a manufacturer, I fill the plate every print, every time with a large number of small objects (up to 500).  When an individual object fails I must have the ability to cancel it so the entire, multi-day print does not fail.

Bug: can’t seem to cancel objects consistently, seems like the larger number objects will not allowed me to cancel and  some of the smaller number objects still allow cancellation.  The error I received through Prusa connect is “HTTP error 422 - command was rejected by the printer” see screenshot below.  In the screenshot I tried to cancel 422 and got the error, however, if I tried to cancel object 200 the cancellation goes through normally.  I’m not sure if it is a number limitation, or if it is a part:plate orientation related issue but I really need this fixed ASAP.  I was able to replicate the issue after cycling power to the printer with the same job.

While you're at it would make a lot of sense if the  object cancellation GUI  was at the top of the screen, above the list of the objects, not randomly three quarters of the way down the list.   obviously this GUI is the only efficient way to identify the part needing to be canceled with a high number of objects on the bed and will be used each and every time a cancellation is needed.

 thank you for the great work  on everything you have done with the XL!

### How to reproduce

Fill the plate with small objects, try cancel them

### Expected behavior

This is a critical issue for those of us that use these in farms.  Please increase limit to 500 objects. I've been using cancel object with 0.0 issues on my entire Prusa farm for years with RPI and octoprint, now that im forced to used PrusaConnect with the XL, I need it work.  Bigger printer, bigger cancel object requirements... Come on guys!

oriddlero, 2025-01-09:
Nobody cares?

CZDanol, 2025-01-09:
Hello, we do care.

There is already the issue #3646 regarding this and we have an internal ticket BFW-6465 for this. This issue is looking at the thing from slightly different angle, so I will not close it as duplicate.

You can currently cancel only first 64 objects (and unfortunately only first 16 from the menu). I don't believe that cancelling even object 200 would do anything.

I'm sorry to say but while I am trying to make this happen, it will likely take some time.

KERZAS, 2025-01-09:
I second with the issue, same situation, I am a production engineer, we do print parts for manufacturing and in large quantities, full bed. But only different is that I dont get any code or message when canceling the object, it just dont respond and printing continues. Also ill add to this that cancel object function on the printer it self is also not fully functional as it does not show instance number and only fits something like 20 objects on the list in stead of, this time, 245. I did not used "cancel current" option yet. Will update maybe tomorrow if that works or not when printing in large quantities. 
[image]

oriddlero, 2025-01-09:
Thank you for chiming in with the second...  That plate looks like what we do every single day,   I hope they prioritize the web-based GUI,  I just want to right-click and cancel object like I do with Octoprint on all of my older Prusa Printers.  I don't really care about doing it from the tiny screen on the printer itself, that should be a secondary milestone with fixing this in my opinion.

oriddlero, 2025-01-20:
Please give us updates on this!   I have 2 XL printers printing 450  parts per plate and it is an absolute nightmare when one of them becomes unstuck!!!!!

BerenV, 2025-03-28:
Don't appreciate oriddlero's attitude and manners both here and on the forums, but I would like to say that I too have wasted a lot of filament before I realized that this feature doesn't work right. I'm printing 140 objects on a MK4S and need to cancel seven more objects in the bottom right corner, but it just so happens that they're numbered higher than 64 so I have to scrap the whole print.

Thanks @CZDanol for working on this.

[image]

oriddlero, 2025-03-28:
Hey BerenV,

Thanks for adding your example.

Sorry for my salty altitude, however, If this were a just feature request I would be nice and calm, this is clearly a **bug** as listing all of the objects as cancelable and even illustrating them in Prusa connect in a visual with the ability to cancel and it not working makes it a bug.   A **very** expensive, **very** frustrating bug when you operate a print farm.  It makes it even more frustrating that I have operated a farm for years using Octoprint  with  seemingly unlimited cancelable objects only to "upgrade" to Prusa XL Printers, and Prusa connect, only to lose critical, expected functionality by surprise.  Crazy stuff.

Anyhow, you might want to also read this the thread which was merged Into this one and contains the latest post from the Devs

https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3418#issuecomment-2704339312

BerenV, 2025-03-29:
Hi Oriddlero,

I can definitely see where you're coming from. Thanks for the link to the other issue. I like the temporary workaround of merging parts so you can still cancel by groups. And it's good to hear that the devs are working on getting it in the next major update.

closed 2025-11-24 by github-actions[bot]