prusa3d/Prusa-Firmware-Buddy #4378: [BUG] Filament runout hangs or doesn't feed filament if object is canceled

danukefl, 2024-12-30:
### Printer model

Mk3.9S

### Firmware version

6.2.0-alpha2

### Printing from...

USB Drive

### Describe the bug

When printing multiple objects, if one object is canceled and later in print, the filament runout triggers a filament change, at least two issues occur. I've had both a failure to detect the filament is unload, but also where it won't feed the new filament (both during first load and purge more options).

I've tested:
- two different prints 
- with two different objects
- different USB drives
- runout before canceling an object works

### How to reproduce

1. Start print with 2 or more objects
2. Cancel one of the objects
3. Cut filament off and let it trigger the filament runout sensor
4. Now either filament the filament won't detect the unload or won't feed the new filament if it does detect the unload.

### Expected behavior

Expected to be able to perform a filament change as normal.

CZDanol, 2025-01-08:
Hello, thanks for the report. Created an internal ticket BFW-6578.

3d-gussner, 2025-11-19:
Please try firmware [6.4.0-RC2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.4.0-RC2) and report back.

danukefl, 2025-11-23:
Tested successfully, thanks @3d-gussner and the others involved in fixing this.

closed 2025-11-23 by danukefl