prusa3d/Prusa-Firmware-Buddy #4379: [FEATURE REQUEST] Let users manually overide "incomplete" calibration state

p27182, 2024-12-31:
### Printer model

XL

### Describe the feature

I want the printer to stop prompting me with a modal notification that not all calibrations are done because I didnt complete the filament sensor cal process.  It is not necessary to complete the filament cal process if you dont use it.  (In my case, the printe rhas to be removed from its enclosure to run the filament cal process).

### Expected functionality

prompt for ack once and dont ask again, or let user manually check cal processes as "complete".

schwiftyp, 2025-01-02:
Yes please, it is annoying as I have a lot of MK3.5's with custom mods that cant finish some calibrations unless I do some things to make it believe its stock to avoid these messages, having an off switch would have saved me a lot of hassle.

CZDanol, 2025-01-06:
Hello, we will discuss this with our product department.

p27182, 2025-03-10:
> Thank you for your contribution to our project. This issue has not received any updates for 60 days and may be considered "stale." If this issue is still important to you, please add an update within the next 7 days to keep it open. Administrators can manually reopen the issue if necessary.

I would not have brought this issue to your attention if I felt it wasnt worth resolving.   I have to take my enclosure apart every time an update to firmware requires me to recalibrate and I find that really inconvienient 

thanks

CZDanol, 2025-03-10:
Hello, we have discussed this with the product department. Their statement is that we cannot do this for legal reasons (basically to cover our bums) - so I am sorry, but we will not be implementing this.

closed 2025-03-10 by CZDanol

p27182, 2025-03-11:
> Hello, we have discussed this with the product department. Their statement is that we cannot do this for legal reasons (basically to cover our bums) - so I am sorry, but we will not be implementing this.

well that is a bummer but thank for giving me an explanation at least.