prusa3d/Prusa-Firmware-Buddy #4381: [QUESTION]

isulpt, 2025-01-02:
### Question

I try to compile the 6.1.2 firmware for prusa mini, but have all time this error... It's a bug or i need something special to do?

I have installed all the dependencies, update python, update pip and all time the same error with this command:

python utils/build.py --preset mini --build-type release --generate-bbf

Thank you

-- Build files have been written to: C:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/build/mini_release_boot
[0/3] Generating include/esp_parts.gen
[2/3] Building CXX object src/buddy-esp-serial-flasher/CMakeFiles/buddy-esp-serial-flasher.dir/esp_flash.cpp.obj
FAILED: src/buddy-esp-serial-flasher/CMakeFiles/buddy-esp-serial-flasher.dir/esp_flash.cpp.obj
C:\Users\TPad\Desktop\firmware\Prusa-Firmware-Buddy-6.1.2\.dependencies\gcc-arm-none-eabi-13.2.1\bin\arm-none-eabi-g++.exe -DBOARD=BOARD_BUDDY -DBOARD_VERSION_MAJOR=1 -DBOARD_VERSION_MINOR=0 -DBOARD_VERSION_PATCH=0 -DHAS_MARLIN=1 -DMARLIN_DISABLE_INFINITE_LOOP -DMD5_ENABLED=1 -DMOTHERBOARD=1823 -DPRINTER_CODE=12 -DPRINTER_MODEL=\"MINI\" -DPRINTER_SUBVERSION=0 -DPRINTER_TYPE=2 -DPRINTER_VERSION=1 -DPROCESS_CUSTOM_GCODE -DSERIAL_FLASHER_INTERFACE_UART -DSERIAL_FLASHER_WRITE_BLOCK_RETRIES=3 -DSTM32F4 -DSTM32F407xx -DSTM32F4xx -DSTM32GENERIC -DUSE_HAL_DRIVER -D_EXTUI -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/lib/esp-serial-flasher/include -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/lib/esp-serial-flasher/private_include -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/include -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/src -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/build/mini_release_boot/src/resources/include -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/src/buddy-esp-serial-flasher -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/lib/Drivers/STM32F4xx_HAL_Driver/Inc -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/lib/Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/lib/Drivers/CMSIS/Include -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/lib/Drivers/CMSIS/Device/ST/STM32F4xx/Include -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/include/stm32f4_hal -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/include/device/stm32f4 -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/include/usb_host -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/include/usb_device -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/include/marlin -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/include/freertos -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/src/hw -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/src/common -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/src/persistent_stores -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/src/persistent_stores/store_instances -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/src/guiconfig -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/src/lang -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/src/common/utils -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/build/mini_release_boot/include -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/include/buddy -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/src/common/fanctl/. -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/src/marlin_stubs/MINI/. -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/lib/Middlewares/Third_Party/FreeRTOS/Source/include -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/lib/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/lib/Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/lib/Segger -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/include/segger -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/src/segger -IC:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/src/logging --specs=nosys.specs --specs=nano.specs -O3 -DNDEBUG -std=gnu++20 -Wall -Wsign-compare -ffile-prefix-map=C:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2=. -Wno-register -Wno-volatile -fno-rtti -g -Os -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -MD -MT src/buddy-esp-serial-flasher/CMakeFiles/buddy-esp-serial-flasher.dir/esp_flash.cpp.obj -MF src\buddy-esp-serial-flasher\CMakeFiles\buddy-esp-serial-flasher.dir\esp_flash.cpp.obj.d -o src/buddy-esp-serial-flasher/CMakeFiles/buddy-esp-serial-flasher.dir/esp_flash.cpp.obj -c C:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/src/buddy-esp-serial-flasher/esp_flash.cpp
C:/Users/TPad/Desktop/firmware/Prusa-Firmware-Buddy-6.1.2/src/buddy-esp-serial-flasher/esp_flash.cpp:33:10: fatal error: esp_parts.gen: No such file 
or directory
   33 | #include <esp_parts.gen>
      |          ^~~~~~~~~~~~~~~
compilation terminated.

CZDanol, 2025-01-02:
Hello, all our devs are either running Linux or building on WSL, so I'm sorry, but we don't have anyone to help you with making this run directly on Windows. I would recommend you installing WSL Ubuntu - that one is tested and used.

Some tips/things to check:

1. Have you run the bootstrap script?
2. Are you activating the python venv?

isulpt, 2025-01-02:
> Hello, all our devs are either running Linux or building on WSL, so I'm sorry, but we don't have anyone to help you with making this run directly on Windows. I would recommend you installing WSL [...]

[image]

Thank you for the fast reply.

I just installed WSL with Linux Ubuntu and managed to compile without any major issues. 
Thanks again!

isulpt, 2025-01-02:
[image]

Thank you for the fast reply.

I just installed WSL with Linux Ubuntu and managed to compile without any major issues.
Thanks again!

closed 2025-01-02 by isulpt