prusa3d/Prusa-Firmware-Buddy #4384: [FEATURE REQUEST] Acknowledge the acoustic signal without resetting, in the event of an error

U-B-Glade, 2025-01-02:
### Printer model

MK3.9

### Describe the feature

In the event of an error, the beeper emits a continuous sound. Allowing the issue to be analyzed calmly, it would be helpful if the beeper could be silenced without needing to reset the printer.

CZDanol, 2025-01-03:
Hello, what error do you mean specifically? BSOD? Generally, you can
silence the beeping by turning/pressing the knob.

U-B-Glade, 2025-01-04:
Hello , I have an Overcurrent Error (False Number but is another issue #4383 ) and if I try to figure out, what was the Problem I try to Acknowledge the Signal by Preessing the Button, but that triggered a reset. The Printer Reboot and the Error occurred again with the PEEP. 
[image]

CZDanol, 2025-01-04:
Okay, and rotating the knob does not silence the error?

U-B-Glade, 2025-01-05:
Oh, I didn't even think about rotating. This was counterintuitive to me.

CZDanol, 2025-01-06:
🤔 It might make sense to not reset the printer on knob press, but require the reset button press. I've created an internal ticket BFW-6539 so we think it over a bit.