prusa3d/Prusa-Firmware-Buddy #4385: [FEATURE REQUEST] Allow "Cancel" in addition to "Retry" in case of error.

U-B-Glade, 2025-01-02:
### Describe the feature

In the event of an error, such as when the filament cannot be retracted to the Finda, it would be helpful to have an option to "Cancel" instead of only "Retry," allowing access to the main menu to perform actions like moving the axes.

coasterboy101, 2025-01-21:
I would like to add my support to this request, there have been many times where I have had to reset the printer to get out of an unrecoverable loop which could be avoided if there was a "cancel" button.

LarGriff, 2025-01-29:
Please!

DRracer, 2025-01-30:
MMU operations are atomic from the printer's perspective and they must be accomplished. Currently, there is no way of leaving them half finished or cancelled. This is vital especially while printing, because next G-code can be processed once the MMU finished its job.

@coasterboy101 Unrecoverable loops while printing basically mean you had to scrap the print, so there is no difference to resetting a printer IMHO.

@LarGriff Are you asking for a "Cancel" option during a print? Whats outcome do you expect out of this option? Printing without filament? Scrapping all the other filament changes from now on? I don't see that as a user friendly option, or I don't understand your intent. Please expand.

However:
- you can perform any HW changes while the MMU is standing in an error - I'm wondering what kind of errors you were trying to solve and which have been "unrecoverable"? In the original report, @U-B-Glade mentioned a scenario when filament couldn't have been retracted to FINDA. That is a recoverable error and after freeing the filament from whatever had held it, the MMU will look at its sensors (FINDA and FSensor) and then pull filament back to FINDA.
- performing operations through printer's menu could be made cancellable, actually there is one which is already cancellable (Preload to MMU). Then, it's the user's sole responsibility to sort out inconsistent state between the MMU and the printer - which might lead to other "feature requests" and "bug reports".

LarGriff, 2025-01-30:
I’m asking for an option to Cancel whenever the ONLY option currently is to Retry.  Retrying may or may not solve the problem.  When it doesn’t, I’d prefer to gracefully Cancel the print rather than do a Hard Reset or Power Cycle.

LarGriff, 2025-03-29:
Prusa isn’t listening.  I’ve found a half dozen other requests for the same thing, but all are ignored.  Prusa thinks they know what we need better than we do.

closed 2025-06-05 by github-actions[bot]