prusa3d/Prusa-Firmware-Buddy #4387: [FEATURE REQUEST] Allow user to disable heat absorption/soak

p27182, 2025-01-03:
### Printer model

XL

### Describe the feature

I would be very grateful if I had the option to disable/enable the heat absorption step (or better yet, make its state a function of bed section count so it's automatic).  I know it has its purpose but I can't remember the last time I skipped it and it caused issues...  I'm currently iterating changes on pretty small parts and the heat absorption skip is requiring input at the printer pre-print...
thanks for reading

### Expected functionality

Maybe if 5 or more bed tiles are being heated, then go through the heat soak, but if 4 or less then skip automatically?

CZDanol, 2025-01-03:
Hello, you can adjust your starting gcode and remove the wait for preheat - it's the `P` parameter in `G29`. Removing it will skip the step.

But be warned, the absorbing heat is rather important. The bed is twisting a bit until the heat transfer stabilizes, so if you do bed leveling before it is stabilized, you might end up with bad first layers. It should be done even if only a single bedlet is active.

closed 2025-01-03 by CZDanol

p27182, 2025-01-03:
> it's the `P` parameter in `G29`. Removing it will skip the step.

thanks for the info, but I see that this documentation (linked from prusa'a docs) suggests that the P parameter is the grid size of the MBL?
https://reprap.org/wiki/G-code#G29:_Detailed_Z-Probe

Looks like there are 7 calls to G29 in the default start gcode including these:
```
G29 P1 ; invalidate mbl & probe print area
G29 P1 X30 Y0 W{(((is_extruder_used[4]) or ((is_extruder_used[3]) or (is_extruder_used[2]))) ? "300" : ((is_extruder_used[1]) ? "130" : "50"))} H20 C ; probe near purge place
G29 P3.2 ; interpolate mbl probes
G29 P3.13 ; extrapolate mbl outside probe area
G29 A ; activate mb
l```

the most complicated being:
`G29 P1 X30 Y0 W[ 300 | 130 | 50 ] H20 C`
does this contain the wait?

thanks!

CZDanol, 2025-01-03:
I'm sorry, I meant to write `G`. I'm not sure how that letter got scrambled on the way from my eyes to the keyboard...

p27182, 2025-01-03:
`G29 G ; absorb heat`

ah I should have noticed that with the comment!  thanks! I will try removing it.