prusa3d/Prusa-Firmware-Buddy #4389: [QUESTION] 6.2.0-alpha2 Firmware - Upcoming advancements?

JGeiger2024, 2025-01-03:
### Question

When will the next release of 6.2 occur and what features are planned?

CZDanol, 2025-01-04:
Hello,
I'm sorry, but we're not a liberty of sharing our release schedules.

I can however say that we do not plan to add any significant new features apart from what we've already done in alpha1 and alpha2 (https://github.com/prusa3d/Prusa-Firmware-Buddy/releases), it would be mostly just bugfixes, stabilizations and performance improvements.

We however have some nice features in works for later releases :)

closed 2025-01-04 by CZDanol

JGeiger2024, 2025-06-24:
Any changes coming to fix issues with filament temperature differences when swapping filament types (MMU3)?