prusa3d/Prusa-Firmware-Buddy #4393: [ENHANCEMENT] Prompt for confirmation before changing filament during printing

kylehakala, 2025-01-08:
### Printer model

MK4S

### Describe the enhancement

While printing with the MK4S + MMU3 (running 6.2.0-alpha2 + 3.0.3), the "Change Filament" option in the Tune menu does not currently prompt for confirmation before pausing a print and initiating the change-filament operation (in the case of the MMU, unloading the filament on the current extruder and ejecting the filament before prompting the user to load new filament to the MMU). 

That non-cancelable action feels particularly disruptive especially given the first position on the Tune menu—right beneath the Return button—where it can be inadvertently chosen when using the touchscreen or control knob if not careful. 

It would be nice if the Change Filament option could prompt with a notice (example: "Pause printing to replace current filament?"). 

I'm happy to provide more detail to the request if it's helpful. Thanks for your consideration here! 

Cheers, 
Kyle

### Expected functionality

When selecting "Change Filament" from the "Tune" menu during a print job, the action should require user-confirmation before pausing the job and proceeding with the filament change operation.

CZDanol, 2025-01-08:
Hello, good idea :) Created an internal ticket BFW-6577

murk-sy, 2025-01-09:
Related issue with a confirmation for pause/submenu:
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4033

CZDanol, 2025-02-18:
Hello, the feature has been implemented and should go out in 6.3.1/6.4