prusa3d/Prusa-Firmware-Buddy #4394: [BUG] Extruder is going up when Unload Filament is selected

slackista, 2025-01-08:
### Printer model

MK4/MK4s

### Firmware version

Any

### Printing from...

Any

### Describe the bug

Last print eg was ended on 200mm, printer off. Printer on, select unload filament, Extruder is going up to its maximum for no reason. In enclosure this has very bad consequences - failed unload, broken filament, even clog, unable to easy load new filament etc.

### How to reproduce

Finish print, turn off printer, turn on printer, select unload filament, Z is going up.

### Expected behavior

Please disable it, is very annoying, every time I print I have to struggle with this odd behavior.

closed 2025-01-08 by CZDanol

slackista, 2025-01-08:
I would say I is debatable if is good or not. Actually I've talked to people from my local 3d printing community how often they home their senseless printer after print is finished, or before turning it off. None them. Why you even consider homing before turn off? If 1/9999 use cases the nozzle is close to the bed before purge nothing terrible will happen.

CZDanol, 2025-01-08:
I am pretty sure I have turned off my printers with nozzle quite low multiple times myself. Doesn't have to be homing before turning off, that doesn't make sense, but for example the printed piece can be very small.

slackista, 2025-01-08:
After standard finish printing or manual stop printing Z is going up by 50mm.

CZDanol, 2025-01-08:
It's actually 20 mm, which is not enough for the user with the tweezers to comfortably remove the purged filament.

slackista, 2025-01-08:
Simple statement that 50mm will be minimum after stop/finish. Store that value in persistent memory. Value is cleared every print starts so in case of power loss will fallback procedure is proceed.

CZDanol, 2025-01-08:
Even if we've done that, there still might be situations when for example the user turns off the printer while the nozzle is down. No. There must be at least some minimal liftoff.

We could potentially tweak it and make the filament unload liftoff say 1 cm and the print finish liftoff say 4 cm, but honestly I don't think this is worth dealing with unless there's more people interested in this change.

neofir3, 2025-01-09:
I've had a similar thing a few times.
Print finished, then unloaded the filament. Z raised all the way to clean the nozzle with a brass brush. Then let the printer cool down and then turned it off.
Turned it on again the next day, loaded the filament and it rattled heavily against the upper Z holders.
After that I had extreme play in the black Z trapezoidal nuts and had to order new ones.
Why isn't there a short message for filament loading/unloading asking whether you want to at least home?
I'm not really happy with the current state either. But that was the case with MK3S too. So it will probably stay that way.
I try to pay attention to it manually so that I don't end up with broken Z trapezoidal nuts again.

murk-sy, 2025-01-09:
> Simple statement that 50mm will be minimum after stop/finish. Store that value in persistent memory. Value is cleared every print starts so in case of power loss fallback procedure starts.

I think a more reliable way to do it would be to check the height every once in a while during a print (maybe on layer change or something that is already gated by an if to reduce unnecessary compute?) or after every manual move.

If Z < 40 mm -> Set "Low position" variable to true
If Z > 50 mm -> Set "Low position" variable to false

The 10 mm nothing zone prevents having too many memory writes and also helps account for cancellations and individual object completion printing interruptions.

Of course it's not *impossible* that a user wouldn't turn off the printer at a high position and move it extremely low manually, but I think that's unlikely enough. 

On a side note, on the Mini, the long and slow (especially Z) travel also causes a lot of material to be lost and oozed when dealing with a filament runout for example, though that's more a separate issue.