prusa3d/Prusa-Firmware-Buddy #4395: Printbed pushs to strong against the nozzle

CloudStrife1983, 2025-01-08:
### Printer model

XL 5T

### Firmware version

6.2.0 Alpha 2

### Printing from...

Prusa Connect

### Describe the bug

The print bed presses too hard against the nozzle during homing (only the first time on point 0) before the print start. You can really see how the print head pushes upwards and you can hear a slight cracking sound. 

### How to reproduce

Start a Print and wait until the Printhead is Homing. Not every time, but very often. 

### Expected behavior

Normally, the nozzle should only lightly touch the print bed. With a smooth PEI sheet, I would already have a dent at the 0 point.

CZDanol, 2025-01-09:
Hmm, was this happening on older releases?

Razor412, 2025-01-09:
I am also running 6.2.0 Alpha 2. I don't get a crack, every once in a while you can notice a deflection in the print bed when it makes it's home probe. In fact I now have a dimple on my smooth PEI sheet where the homing point is. 

It actually destroyed one side of my PEI. I contacted support and we went through the normal process of figuring out what happened, and as luck would have it everything checked out ok. Never did that level of destruction ever again in probably around 200 prints since.

CloudStrife1983, 2025-01-09:
For me, the phenomenon does not actually occur every time. And it seems as if it disappears for a while after recalibrating the Loadcell until it occurs again. However, it does not occur with a specific toolhead, but it does not matter which toolhead is referenced. So I suspect that it is a software problem. I also find that with Loadcell calibration you have to press relatively hard against the nozzle to perform the calibration successfully. Regardless of the toolhead.

CZDanol, 2025-01-09:
All right, sounds like possibly a thing. I've created an internal ticket BFW-6591 for this. We will try to investigate.

Dragonos2000, 2025-01-10:
I can confirm, that the first probing point leaves a deep bump on the PEI sheet and it's at least visible on the other surfaces. However, I'm using 6.1.3 and did not yet test a newer firmware.
[image]