prusa3d/Prusa-Firmware-Buddy #4400: [BUG] Extruder Motor still stops

Gik-0101, 2025-01-10:
### Printer model

XL Multi Tool

### Firmware version

6.2.0 RC1

### Upgrades and modifications

Fan Shroud 360, Enclosure

### Printing from...

Prusa Connect, USB

### Describe the bug

Extruder motor still stops at the layer height where the component fan switches on. 
Mainly affected tool head 2
Dwarf cover connector has already been removed (recommended by support).

It doesn't matter which material is printed.

Thanks to FW 6.2.0 RC1 now i get a "red screen" and the Printer stops.

[image]

### How to reproduce

Not reproducible

3Dwyer, 2025-01-14:
I'm having the same issue. Firmware did the same thing for me as well. Just gives a red screen that extruder stopped. Support sent me 2 new extruder motors and 2 new extruder cables. They had no affect on the outcome.

balloonhotair, 2025-01-15:
I just installed this version of the firmware and got the same error message 4 hours into a 12 hour job. What is worse now is the error message is the end of the job. In the previous version if I noticed this bug I would pause the XL and hit the nextruder reset button on the side of the toolhead and once the rest was done I could resume the job and the extruder would keep going. I would like to see them do the same in the firmware as a workaround while they try and find out what is causing this bug.

CloudStrife1983, 2025-01-15:
In my opinion, the various errors such as: Extruder stops, Nozzle LED flickers strongly and also the Min Temp errors that suddenly occur at work, which occur sporadically on 3 of the 5 extruders without forewarning, are all related. The contacts of the plugs simply deteriorate due to a lot of movement, which leads to the problems mentioned. Prusa absolutely must make improvements here. This is simply a design error. The plug needs a reasonable and reliable strain relief.

Gik-0101, 2025-01-15:
I've been bothered by this problem since August (I think) 2024.
I have the normal pulsing of the nozzle LED when heating up and holding temp., but if the extruder fails it strobes very quickly.
But I've never had the minimum temp. problem.
Without starting a technical discussion here, I don't know that much about the XL. But I think the problem lies deeper than a plug connection. I couldn't trigger the error by moving the plug connection while printing. It shouldn't be purely a software problem either.
Anyway, I just want a solution slowly.
That's why I opened the error here.

balloonhotair, 2025-01-15:
Yes I agree with everything you said above. I have looked at the code changes that the developers made in the last release to address this issue and at a high level they are dealing with race conditions with locking semaphores. It is what can be called a starvation problem and when it happens the current code has no recovery way out except for now putting up an error message, which has the comment in the code "Never should get here". LOL So this issue is probably both a hardware issue and a software issue since it is so "random".  Like I said above instead of the error message the printing should pause and then a reset of the ARM processor should be done and then a resume, at least this would let then print continue. IMO.

Gik-0101, 2025-01-15:
Yes, that would be a sensible and intelligent interim solution to avoid always producing incorrect prints and wasting material.

bkerler, 2025-01-15:
My best guess from my experience is that this isn't a software issue, but hardware issue. If the error occurs randomly, it might be PSU or XBuddy Board related. Check all cables for broken shielding and also that all connectors are connected properly. If all that is fine, I'd recommend to replace the main board.

jxtan-msn-com, 2025-01-15:
> I have the normal pulsing of the nozzle LED when heating up and holding temp., but if the extruder fails it strobes very quickly.
Just got hit with this issue, and I also noticed that the extruder LED light pulsing. It was working for awhile, then new print, this happens. And after stopping the print, the purge filament won't work either until I reset the XL.

balloonhotair, 2025-01-15:
Just FYI if you notice this when printing and not using this RC version of the software you can pause the print from the console and hit the reset button on the left side of the extruder. This will reset all the extruders to blue and then green color. Then a resume on the console will start the print again from where you paused it.

CloudStrife1983, 2025-01-16:
> My best guess from my experience is that this isn't a software issue, but hardware issue. If the error occurs randomly, it might be PSU or XBuddy Board related. Check all cables for broken shielding [...]

These are all things I have already done. I have already installed a new sandwich board. Nothing had helped. Simply reconnecting the plugs solves the problem of the extruder stopping for quite a while for me. However, the flickering of the nozzle light remains very minimal and becomes more intense the more I print.

jxtan-msn-com, 2025-01-16:
On the facebook group, someone said it is related to bad extruder cable.

> These are all things I have already done. I have already installed a new sandwich board. Nothing had helped. Simply reconnecting the plugs solves the problem of the extruder stopping for quite a [...]

On the facebook group, someone said after working with support, that specific case is related to bad extruder cable.

Gik-0101, 2025-01-16:
IF that's true, hopefully those of us affected will hear something from support soon.
My support just told me yesterday that some things are still being tested internally. But it will soon be completed.
Let's see, I've already been promised a lot in this direction.

balloonhotair, 2025-01-17:
Well I finally have a workaround that for me seems to prevent the extruder motor stalling. I have a job that I have tried to print almost a dozen times in the last month or so and it always fails with one of the extruders not turning. I also had the flashing led issue but it was not my focus since the extruder motor stalling was a bigger problem. I have now run the 12 hour job twice without any issues and the flashing led is no longer an issue as a bonus. I created my own strain relief for the cable and connector. It is crude and I suspect Prusa will have a nicer way of doing it soon. I attached a picture of what I have and hope it will help someone else.

[image]

CloudStrife1983, 2025-01-17:
> On the facebook group, someone said it is related to bad extruder cable.
>
>
> On the facebook group, someone said after working with support, that specific case is related to bad extruder cable.

He was probably just told by support to "swap cables". After that, it's always gone for the time being. As long as there is nothing official from Prusa about this, it's just hot air for me. @Gik: Many people in the forum have already swapped cables and it didn't help, if I'm not mistaken.

CloudStrife1983, 2025-01-21:
I'm currently testing this mod here: https://www.printables.com/model/1124941-secure-xl-nextruder-cable-connector-solution
This seems to be the solution to the problem, at least for now. Of course, I'll have to print for a while to see if it really stays that way, but at the moment it seems to me that the problem is caused by loose plug connections or plug connections that come loose after a while. In other words, exactly the opposite of what Prusa is currently stating.

peterjhl, 2025-01-24:
Updated printer maintenance and new parts - https://help.prusa3d.com/guide/how-to-replace-the-main-cable-connector-cover-xl_831370   --  https://www.printables.com/model/449554-xl-printable-parts/files#preview.file.Zxvo4  -- https://www.printables.com/model/449554-xl-printable-parts/files#preview.file.K7TdZ  -- https://www.printables.com/model/449554-xl-printable-parts/files#preview.file.IG9Uh

TheYang, 2025-01-24:
I've had aborts on two different new Extruders as well. Happened in two of the last 3 prints. I think in the first one (the only one that worked) I only used older Extruders.  

I think the intermittent solution (for Firmware) may be to simply log short disconnections, but Ignore them (assuming ignoring them was what happened before 6.2 rc1). Long term, of course the whole disconnection issue would have to be sorted.

Gik-0101, 2025-01-24:
@TheYang

TheYang, 2025-01-24:
@Gik-0101 No, sorry.  
It's just that I got an 2TH late 2023 and upgraded to 5TH late 2024, now after the update I have issues, and the issue happened on the late 2024 extruders.  

I did have (seemingly) random pauses and filament unloads before FW 6.2 rc1, but never an aborted print from missing extrusion from what I could tell.

CloudStrife1983, 2025-01-25:
I have tried to install the new cover. However, it's way too tight of a tolerance and it either breaks apart when you assemble it, or you get the hose and cables caught in the top screw when you try to tighten it. I have printed this in PDF and it is simply impossible to assemble it properly. The tolerances are far too tight.

CloudStrife1983, 2025-01-25:
Is it also possible to make the STEP file available at Printables? Currently only the stl is updated, the STEP file is the one from the old connector cover

CloudStrife1983, 2025-01-25:
For everyone who print the cover in PCCF. Print it with Extrusion Factor 1 instead 1.04. Installing the cover without the jig is more easy with the following steps: 
First of all put the Toolhead on the toolchanger and pull it to the front 
1. connect the cable on the extruder wihout the cover. Lay the cable straight up, as it should look in the end. 

2. Open the cover slightly as shown in the instructions and insert the nut

3. Slide the cover onto the cable from the left, press it together carefully with your hand, taking care not to pinch the cable at the screw at the top. 

It is important that the sleeve only goes in the upper area where the hatching is. Not all the way down. In other words, as shown in the official instructions. 

4. Fit the two lower screws but do not tighten them yet. Only until they lie flat. 

5. Make sure that the cable sleeve and the cable are not pinched and tighten the screws. 
6. Finally, tighten the two bottom screws

Gik-0101, 2025-02-03:
I have now installed the modified Prusa parts and have already done a few prints with tool changes. 
So far all have been successful and no extruder failure has been recorded yet.

So far, I can recommend the official conversion to anyone who is affected by the extruder failure problem.

If anything changes about the problem I will pass it on.

JPElectron, 2025-02-06:
So after updating to firmware 6.2.0 I tool experienced this problem randomly on Tool head 1 and 2, couldn't finish any print, so I printed the updated parts to "keep the connector in" and I still had the problem, randomly on Tool head 1 and 2.  I reverted back to firmware 6.1.3 and have now had 6 successful prints without issue.  How am I to conclude this is a hardware issue??

CloudStrife1983, 2025-02-06:
> So after updating to firmware 6.2.0 I tool experienced this problem randomly on Tool head 1 and 2, couldn't finish any print, so I printed the updated parts to "keep the connector in" and I still [...]

The contacts have probably already deteriorated too much for the new covers to solve the problem. In my case, the nozzle LEDs still flicker at TH1,2,5, but it's not getting any worse now. Maybe you should write to support and replace the dwarf circuit board and the cables as well.

Gik-0101, 2025-02-06:
My nozzle LED also still flickers. But with the conversion kit I still didn't have any extruder failure.

But I'm still not sure whether this has finally been solved and can be checked off.
Only time will tell.........or a new FW.

CloudStrife1983, 2025-02-08:
> My nozzle LED also still flickers. But with the conversion kit I still didn't have any extruder failure.
> 
> But I'm still not sure whether this has finally been solved and can be checked off.
> Only time will tell.........or a new FW.
> 

However, I have noticed that the flickering gets stronger again over time on my XL.

Nepharoel626, 2025-02-08:
I have now similar issues like everyone here it seems, after updating my 5T today and the here common error for toolhead 1 occurred mid print (approximately 10 layers). Note that nozzle fan settings are: "Off until the 6th layer" and "full speed allowed after the 12th layer"

The nozzle LED has occasionally flickered for a long time now (over several FW Versions) and also sometimes the left LED Strip. In addition I had occasionally problems with the "Fan Test" (1st calibration), but since the issues were as mentioned before occasionally or rather randomly, as well as only the "Hotend fan RPM test" from toolhead 1 sometimes did not pass the test and the problems of absurd layer shift (printing error or aborted print without possibility to resume) 5,5h into a 7h print seemed to be solved after a extended period of troubleshooting with the Prusa support and FW Updates, I did not put much thoughts into the minor LED and Fan Test issues.

However, now I get a similar error message as in picture by @Gik-0101 shown above in the starting post for my toolhead 5, when I try to run the fan test and could not make it passed ever since. Because as soon as I hit the button, without even showing the test screen for a millisecond, the system goes to restart and after the "Bootloader" screen the error shows up, which seemingly pains us all currently. Moreover is it strange for being about toolhead 5, which was since I got my XL in April 2024, was never in use yet a long with toolhead 4 and thus was never moved besides calibration or maintenance. 

Ultimately, I would like to ask if someone could provide any help/suggestions to fix it momentarily to be able to print the the official conversion parts (if still recommended as fix) or what would be your best bet for a problem description towards prusa support to avoid another long lasting journey of "did you tried this or that" according to your experience with this issue (e.g. could I just refer to this thread here)??

BillC58, 2025-02-14:
I too have updated to 6.2 and installed the new cable connectors. The extruder error is still happening to me. 

Perhaps the cables need replacing. I will try that next.

Gik-0101, 2025-02-15:
@Nepharoel626 

Sorry for my late reply.
Normally if you photograph the red screen with the extruder stop error message and send it to Prusa support there is no longer a "have you tried that" response. Here they are now understandable.

You can temporarily bypass or delay the error by unplugging the cable on the Dwarf board until the XL reboots and then plugging it back in. Reset the printer when it is in the red screen and the error should no longer occur for 2-3 prints. SHOULD! I can't promise it, it worked for me.

If you install the conversion kit as described by Prusa (exactly as they describe it, not sloppy) the error should go away. I haven't had it anymore until now. It's been almost a month now and I do print a few things but not constantly or permanently.

good luck

Prusa-Support, 2025-02-21:
Thanks for creating this issue. We will keep an eye on it.

This is a sporadical problem that is known at the current time, so a specific error has been added to the firmware.
Several aspects that may potentially lead to this problem have already been addressed but there may be additional factors for making considerations.

- One of the addressed concerns is data clogs on the Dwarf board for electronic reasons due to a partially loose Nextruder cable.
Please try to temporarily remove the 'dwarf-cover-connector' and check if that helps.
https://help.prusa3d.com/article/extruder-error-17536-xl_822367
  - If so, you may want to print and replace the new printable parts 'dwarf-cover-connector' + 'dwarf-cover-base'.
Ideally, please go through the instructions in advance and check the required small fixings before starting.
https://help.prusa3d.com/guide/how-to-replace-the-main-cable-connector-cover-xl_831370
- If the problem persists, we may be looking at hardware problems with the Nextruder cable, the Dwarf board, or the Sandwich/Splitter board... _but that may not be all_.
  - In the case of multi-tool version, swapping parts between tools/docks may help to narrow the problem down to specific components but you are strongly advised to please contact our Customer Support to get help.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research

rovili, 2025-03-01:
I want to share my story with the Prusa XL 5 heads:

**Phase 1:**
- I got a Prusa XL on the 13th of February. It was delivered with 6.1.3.
- I did a small 6 hour multicolour print with 5 colours, everything went fine. Perfect print.
- During the print I stumbled on the new Firmware 6.2.0 and flashed it after the first print was finished and started the same print again, it failed with Error 17536 on E2 after one hour in. So the error occurred after just 7 hours of printing.
- 
**Phase 2:**
- I researched about the issue and found the new cable covers and printed it: https://help.prusa3d.com/guide/how-to-replace-the-main-cable-connector-cover-xl_831370
- Started the same print again with the new dwarf cable covers and again, Error 17536 on E4 after on hour into the print
- I concerted Prusa support, which sent me 5 new dwarf cables
- While the cables are on its way to me, I made more tests, excluded the failed Extruders and eventually all Extruder have failed with Error 17536
- 
**Phase 3:**
- I got the new dwarf cables and replaced the cables
- Firmware 6.2.1 was out, so I flashed it
- Started the same print, again after one hour in, Error 17536 on E3
- Did some more testing, again excluding the failed Extruders and eventually all Extruder have failed with Error 17536
- I printed more things, but only single color, prints between 8-16 hours, one print with each Extruder. Not a single Error 17536
- Contacted Prusa support, documented the whole story -> Prusa sends me 5 new dwarf boards - not arrived yet
- Checking the Power supplies voltage during a print, all fine - was a long shot suggestion by Prusa support

**Phase 4:** while waiting for the new dwarf boards to arrive and being super frustrated 
- Replaced USB stick with a high quality Kingston 
- Started a 5 color print, 1 hour in, failed Error 17536. Started again to exclude the failed extruders in the progress, until eventually all extruders have failed
- I figured, let's downgrade to 6.1.3 and start the same 5 color test print. It's a print with 4000 swaps, each layer has 5 colors
- The print was successful, 16 hours. No missing layer, no artefacts or whatever, just a perfect print

**My findings**
- No Error 17536 with Firmware 6.2.x with single color prints
- All extruders failed with Error 17536 with Firmware 6.2.x when printing multi color
- Multi-color prints are perfect with Firmware 6.1.3
- Replacing the dwarf cables with new ones, does not change anything. Also kinda obvious, since the printer is brand new
- Error 17536 only occurs when all Extruders are parked - meaning, it goes into 17536 when is docking the used extruder and switching to next one

**My concolusion**
- I am 99% sure that is is no hardware-issue in my case
- I am 99% sure it's a firmware issue - yes, I am aware that Error 17536 was just build in with 6.2.x

@Prusa-Support please fix this issue :(

Gik-0101, 2025-03-07:
After many successful hours of printing, the extruder stop error is back.

Nothing has been changed on the tool heads since the recommended conversion kit.

Very annoying that this still hasn't been fixed.

[image]

Parra87, 2025-03-13:
I would like to add that I have also been experiencing this issue, despite upgrading the offending toolhead with the new cable strain relief cover.

[image]

CloudStrife1983, 2025-03-14:
Replacing the extruder cables and dwarf boards doesn't seem to make it any better. So I wonder why they keep sending people new hardware, which doesn't seem to be the problem. Please Prusa finally find the error and fix this annoying problem.

Prusa-Support, 2025-03-14:
Please note that the new printed parts are only part of the solution.
The extruder cable may be fatigued or faulty... or the Dwarf board, or the Sandwich board, or the Splitter board... or even, rather unexpectedly, other pieces of hardware.

We can't exclude other faulty or damaged pieces of hardware or damages during manipulations of even new parts _**for now**_.
Please contact the Customer Support _in all cases_ before further escalation on GitHub, following the guidelines - https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4400#issuecomment-2674851235.

At the current state of the investigation, the problem is not easily reproducible on all printers except for hardware problems.

@rovili Because you claim that most of the parts have already been replaced - not all potentially concerned parts, but many - feel free to write back to the Customer Support to my attention, including a link to this GitHub comment, so I can monitor your case more closely and consider further escalation with the developers, or follow-up with the assigned agent though the natural escalation of your case with the help of product specialists from various departments.

In all of the other cases, please contact our Customer Support and continue the conversation if new printable parts or a cable/board replacement is not sufficient - it is in most cases, but we may be looking at several concerned hardware parts.
_We pledge to solve all remaining cases on an individual basis._

Michele Moramarco
Prusa Research

rovili, 2025-03-14:
@Prusa-Support Thanks, I am in contact with Blas, since the replacement of the dwarf boards and XL splitter did not resolve the issue.

CloudStrife1983, 2025-03-14:
> Please - please make this error into a print pause in stead of a hard error. I'm ok it being a hardware error and you need to get to the bottom of it. I'm ok having it. These things happen. I'm not ok you wasting our filaments while it can be avoided. This will make people revert back to 6.1.x where you will have less chance to resolve the issues.

I have to agree with this. It would be very good if you could continue printing as soon as the error occurs. It often helps to reset the printer so that the error disappears temporarily and you can continue printing, which should actually be possible without any problems using the Powerpanic function. However, this is not the case with a red screen error.  I would generally welcome it if it were possible to continue printing by restarting the printer in the event of other red screen errors. Especially when such an error occurs towards the end of a long and large print, it is very annoying because you waste a lot of time and filament.

amishne, 2025-03-15:
Unfortunately I encountered this error as well, on my year-old printer. From this comment thread it seems clear that:

1. This occurs on both brand-new and on year+ old printers, suggesting it isn't related to hardware fatigue.
2. In most (or all?) cases, this occurs shortly after updating to 6.2.x - including in my case.

Given the above two items, @Prusa-Support, I don't see how you can possibly disassociate this from firmware version. I'm definitely going to revert to 6.1.3 until Prusa's investigation is complete.

I'll report back here if I encounter this on 6.1.3, though!

CloudStrife1983, 2025-03-15:
> This occurs on both brand-new and on year+ old printers, suggesting it isn't related to hardware fatigue.

It occurs on both because the problem has not yet been fixed. This problem has existed for over a year and not just since yesterday.

amishne, 2025-03-15:
> > This occurs on both brand-new and on year+ old printers, suggesting it isn't related to hardware fatigue.
> 
> It occurs on both because the problem has not yet been fixed. This problem has existed for a year and not just since yesterday. My first time with this problem was May 24

That's certainly a possibility, plus [the OP also mentioned](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4400#issuecomment-2593538743) this also happened to them in August 2024, so maybe it was there all along and and firmware 6.2.x just exposed this by reporting an error instead of silently failing.

But (1) I don't remember it happening to me before 6.2.x and (2) the rest of the comments here also suggest this might be a recent issue, so I definitely want to give 6.1.3 a go while Prusa is still investigating, before starting all the troubleshooting steps above that some mentioned did not solve their problem.

danukefl, 2025-03-25:
My XL 5TH assembled was delivered March 2024 and I started getting this issue on all 5 toolheads within a week of one another about a month ago on 6.1.x firmware. I've since updated to the latest RC and receive the errors regularly. I am now hard powering off and on the printer before any large prints and wiggling the connectors and it will work for a random amount of time before it fails again. With all of them happening around the same time tells me its (probably) not the dwarf boards or the toolhead expansion board. I don't have a very balanced print distance across print heads either so I don't think it is wear and tear based, and the printer hasn't been moved since initially setup. My TH1 is a 0.25mm nozzle for fine details on some prints but still throws the error.

I am trying this strain relief at https://forum.prusa3d.com/forum/original-prusa-hardware-firmware-and-software-help/xl-extruder-stops/paged/15/ to see how it will work.

Print distances:
TH1: 1.3km
TH2: 6.6km
TH3: 3.3km
TH4: 2.5km
TH5: 5.3km
13.6k Changes

Stego72, 2025-03-30:
I suspect this is a "overdetection" problem in the firmware. Previous versions could not detect the error, so code was implemented to be able to.

However, a lot of users who never had the problem are now getting the error (probably without anything actually being wrong with the hardware - especially it also appears on units factory-shipped with the improved connector)

I think the fault detection method is oversensitive and fires "too early" on units not having any mechanical problem.

Having the same problem on an almost unused machine (supplied with the latest connectors) - probably going to downgrade the FW to see if "really happens" or if it is "jumpy" fault detection.

danukefl, 2025-03-30:
I am still on the latest but after about 150 print hours on the tool head that I have the new strain relief that I linked above, I have not had any issues. @Stego72 I would like to think it is an "overdetection", but I am wondering if it is catching the extremely small momentary disconnects and shutting it down, versus just proceeding through the blip.

Stego72, 2025-03-30:
> I am still on the latest but after about 150 print hours on the tool head that I have the new strain relief that I linked above, I have not had any issues. [@Stego72](https://github.com/Stego72) I [...]

Downgraded to 6.2.1 to see if it helps - just had the error on a docked toolhead - what happened was:

Holder docked T1 perfectly 
On it way to pick T5
Error appear on T1 (red screen)

I cannot see how you can get a fatal extruder not spinning error on a docked toolhead

onslack, 2025-03-31:
Just wanted to chime in

I've also had this error originally on T1 , which lost me 12hrs of print progress 

So I moved the filament to T3 which is my least used tool head at the moment and got the same error , losing 8hrs print progress 

So I downgraded the firmware as I knew I didn't seem to have issues on the last one 

I contacted support who recommended that I upgrade back to 6.2.2 and remove the dwarf board connectors, which I did and the print managed to finish 

So I've since re printed the connectors and so far * knock on wood * it's been back to it's reliable self again but I'm going to keep an eye on it as I've got multiple day long prints lined up

ddavidebor, 2025-04-01:
I'm getting the same error again on T1, an older design which has been upgraded to the newer cable holders.
Error appeared to have been triggered when the extruder was docked!

Stego72, 2025-04-02:
> I'm getting the same error again on T1, an older design which has been upgraded to the newer cable holders. Error appeared to have been triggered when the extruder was docked!

Kept getting the same - I did some thinking and I'm testing out what I hope to be a fix for us with upgraded cables - see here: https://help.prusa3d.com/article/extruder-error-17536-xl_822367 (long post by "DinoPrint" approx 11:30 CET 02/04/25)

It's working so far :)

Stego72, 2025-04-03:
> I'm getting the same error again on T1, an older design which has been upgraded to the newer cable holders. Error appeared to have been triggered when the extruder was docked!

The #17536 error on docked tools has now been confirmed as a FW bug - see https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4576#issuecomment-2776290166

Let's try to push for a Hot-Fix

amishne, 2025-04-03:
> The #17536 error on docked tools has now been confirmed as a FW bug - see https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4576#issuecomment-2776290166

This also explains why people report this error on multi-head prints but never on single-head ones.

CloudStrife1983, 2025-04-04:
Same issue here. After 14h print Time. Very nice because all of my filament is now for the trash bin. Covers already chabged. Same Probleme! :( [image]

Stego72, 2025-04-04:
Just got off Prusa Support Chat - instruction was to downgrade to FW 6.1.3 to temp solve issue about 17536 on DOCKED toolheads

Prusa-Support, 2025-04-22:
Please upgrade to final [FW 6.2.4](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.4), which would drastically reduce (or remove) the margin of false positives for [EXTRUDER ERROR #17536](https://help.prusa3d.com/article/extruder-error-17536-xl_822367).

**As a reminder, this error message was introduced in FW 6.2.0.**
Any older firmware version is not recommended, not only because technically obsolete, but also because it would simply ignore problems related to this error.

In either case, if this error appears, please contact our [Customer Support](https://help.prusa3d.com/article/customer-support_2287).
We have solutions for you - https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4400#issuecomment-2724268862.

Michele Moramarco
Prusa Research

closed 2025-05-01 by Prusa-Support

jedisct1, 2025-12-15:
> By cleaning the connectors of the cable-splitter-dwarf with ISO alcohol and [a brass brush](https://www.prusa3d.com/product/prusa-brass-nozzle-brush/) the flicker goes back to extremely minor to no flicker.

Cleaning these tiny fragile connectors with a brass brush feels a little bit aggressive.

Ro3Deee, 2025-12-15:
my solution: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4579#issuecomment-3657225481

Poor man's solution for flickering and extruder not spinning on Prusa XL:

[image]
https://youtu.be/xBL0WHxMq0k?si=p9UlQogPvdhqr_qH