prusa3d/Prusa-Firmware-Buddy #4401: [BUG] 6.2.0-RC1   on MK4S resets to 'not HF nozzle' after flashing

beng27, 2025-01-10:
### Printer model

MK4S (upgraded from MK3S, kit build)

### Firmware version

6.2.0-RC1 

### Printing from...

NA

### Describe the bug

flashed with 6.2.0-RC1 ; first print complained that HF Nozzle not installed. Went to settings and it was turned off - it had been on before flashing, 

### Expected behavior

flashing should preserve status of HF-nozzle flag

Zode, 2025-01-31:
Can confirm. Updated to 6.2.0 from the website, imagine my confusion when my printer suddenly told me I didn't have a stock HF nozzle 😂

CZDanol, 2025-01-31:
Whoops, sorry guys, this issue got through the cracks :/

danopernis, 2025-02-05:
Thank you for reporting, this is quite serious. We are working on fixing this. For folks who already upgraded, the only fix is to manually configure high-flow nozzle.

CZDanol, 2025-02-05:
Internal ticket: BFW-6727. We have thought of a way to hopefully fix this even for the guys who already upgraded.

Prusa-Support, 2025-05-20:
Starting from firmware version 6.2.1, _MK4S printers will default to recognizing the nozzle as high-flow_.
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.1

Please allow me to close this issue as solved.

Michele Moramarco
Prusa Research

closed 2025-05-20 by Prusa-Support