prusa3d/Prusa-Firmware-Buddy #4402: [BUG] Firmware file within folder not detected

me25542, 2025-01-11:
### Printer model

MK4S (irrelevant?)

### Firmware version

6.1.4 -- 6.2.0-RC1

### Upgrades and modifications

Custom enclosure connected via I2C

### Printing from...

USB

### Describe the bug

When updating from firmware version `6.1.4` to `6.2.0-RC1`, I discovered that firmware (`.bbf`) files located in a folder (I put the new firmware file in a folder called `Firmware` on the USB) are not detected or flashed. Moving the same firmware file from the folder and to the root directory of the drive resulted in it being discovered and flashed (with confirmation).

### How to reproduce

1. Load a printer with some non-latest version of firmware (in my case `6.1.4`).
2. Put a newer firmware version (in my case `6.2.0-RC1`) on the USB drive, located within a folder (in my case named `Firmware`, but I doubt this is relevant 😉).
3. Insert the drive into the printer.
4. Reboot the printer.
   - The printer will not prompt for firmware installation.
   - If the firmware file is located in the root directory of the drive, and steps `3` and `4` are repeated, firmware installation is prompted as usual.

### Expected behavior

The printer should detect and prompt for installation of firmware files located anywhere on the USB drive, possibly excepting hidden files or folders.

LoakAudio, 2025-01-11:
Hi, I don't think it's a bug. I think devs want to have a fast detection of a firmware file on the usb key by avoiding a scan of a key with folders and sub folders and sub-sub folders ...
Putting the firmware file at the root of the usb key is a part of the "instruction for use".

CZDanol, 2025-01-11:
Hello, yep, not a bug, that's just how things are.

Not scanning subfolders in situations like this is kind of a standard and changing it would be detremental to the printer boot time.

closed 2025-01-11 by CZDanol

me25542, 2025-01-11:
I see. it would be ideal to have a system that could scan the whole drive quickly, but with the hardware constraints this makes sense.