prusa3d/Prusa-Firmware-Buddy #4404: [FEATURE REQUEST] Filament Weight Tracker

schwiftyp, 2025-01-11:
### Printer model

Buddy Boards

### Describe the feature

A feature where you when load a full spool of filament to the printer you can tell the printer how much it has so it knows how much it has used and relays to the user how much is remaining on the spool. A warning for prints that are more in weight of filament than what is remaining on the spool so the user knows it will trigger run out sensor or not during the print and either load new spool or agree to the run out happening and continue.

### Expected functionality

Load filament. While loading the printer gives an optional selection for "Filament Weight" or can be ignored. User inputs how much filament was loaded. Default start of 1kg since its most common. Printer then displays how much filament is left in grams or there is a place in settings to go check how much is left. Warning messeges for gcodes containing weights higher than remaining spool weight.

blue0ceanby, 2025-01-30:
There are already RFID tags inserted into the Prusa spools. When will an RFID reader be available for the printers that could be used for this?

CZDanol, 2025-01-30:
We are currently not inserting NFC tags into the Prusa spools. At this moment, we are using UHF RFID chips that we use for inventory keeping and identifying the spools across the factory. These chips are cool and all, but readers for them cost a lot - so it's only practical for the factory, not for end users.

blue0ceanby, 2025-01-30:
Oh, that's a pity. Perhaps you could adapt the chips to NFC or similar technology and also read them via the printer.

CZDanol, 2025-01-30:
😉