prusa3d/Prusa-Firmware-Buddy #4409: [BUG] Short links in XL IS calibration doesn't exist

RichiH, 2025-01-14:
### Printer model

XL

### Firmware version

6.1.3+7898

### Upgrades and modifications

1T->5T

### Printing from...

USB drive

### Describe the bug

While re-calibrating the Input Shaper after upgrading from 1T->5T over the last few nights, the XL told me to read URLs don't exist.

The text goes to prusa.io/xl-iscal, the QR goes to prusa.io/xl-is**_-_**cal, and neither of the two links exist

[image]

I'd send a patch if I knew what the correct URL should be.

### How to reproduce

Start IS calibration

### Expected behavior

Show me working URLs

D05S, 2025-01-15:
Hello, thanks for the warning, the link should be working now.

closed 2025-01-15 by D05S

RichiH, 2025-01-18:
Confirming both links work now. (I had the QR written down wrong, but it also works now). Thanks!