prusa3d/Prusa-Firmware-Buddy #4414: [BUG] Unexpected thermal runaway error at 270 on MINI

kolba, 2025-01-15:
### Printer model

MINI

### Firmware version

6.0.x, 6.2.0-RC1

### Upgrades and modifications

direct drive mod, silicone sock, copper heatblock, hardened nozzle

### Printing from...

via prusa connect

### Describe the bug

I've been getting suspiciously high amount of thermal runaway orange screens in past 9-12 months. It was not in issue previously as was printing PC and nylon at 285 degrees without an issue, but it seems that since upgrading to Input Shaper firmware, or maybe to 6.x.x I keep getting somewhat random thermal runaway errors. I print alot of recycled PET, and since I want to print faster with hardened nozzle, I have to bump the temp from usual 260 to around 268 (since PET goes white on 270+) and after a bunch of failed prints it seems that in some models, when printing bridges or top of infill layer, which needs 100% fan, the temps dip down and up, hitting 270+ and it just gives orange screen.

I've googled this previously, and it seems that at some point the safe temps were dropped for MINI, so was waiting for a fix. A recent crash reminded me of that, so it seemed like 6.2.0-rc1 had a fix for this, but this still happens it seems.

### How to reproduce

Its somewhat random, but for me:
- print something (eg PET) with 268 temp (prusa slicer, 75 bed, 262 first layer, fan 0%-75 with 100% on bridges, 6.5 max volumetric speed, 80 for walls, 115 for infill)
- your model should have a supported area (7cm x 5cm seems enough)
- rest of the model should be ok, so fan is at 0 most of the time
- when the printer starts printing areas which switch fan from 0 to 100, a temp instability should happen, which could deviate above 270 for a moment, resulting in a thermal runaway orange screen of death

### Expected behavior

Expecting it to work with at least 280 degrees allowing for some short overheating, especially when fan kicks in.

### Files

i've managed to take a quick snap from prusa connect

[image]

closed 2025-01-16 by kolba

nvtkaszpir, 2025-03-30:
Why did you mark it as resolved?

kolba, 2025-03-30:
> Why did you mark it as resolved?

It was something else. Firmware works as expected