prusa3d/Prusa-Firmware-Buddy #4417: [BUG] MMU3 Overcurrent when using USB-C serial port

raistlinJ, 2025-01-18:
### Printer model

MK4S with MMU3

### Firmware version

6.2.0-RC1

### Upgrades and modifications

Prusa enclosure, Prusa GPIO Board, Cable connecting GPIO to Enclosure Basic Board, MMU3, Camera connected to xBuddy board via PSU Power In and adapter, Enclosure LED Strip

### Printing from...

N/A

### Describe the bug

When using the USB-C port on the Prusa xBuddy board with the MMU connected, the firmware reports an MMU Overcurrent error.

### How to reproduce

When the printer is off, connect a USB-C cable from the xBuddy Board to a laptop, and ensure there is an MMU connected and the option for MMU is on. Then, turn on the PSU and the error will occur.

### Expected behavior

The printer should be able to turn on normally with no problems. To resolve this issue, the MMU was disconnected and the printer no longer gave the error.

Prusa-Support, 2025-03-12:
Thanks for reporting. 
The problem seems to be hardly reproducible, but we have been investigating several MMU Overcurrent scenarios.
This specific one, may potentially be unrelated to many others we have been studying so far, but we would appreciate you testing [FW 6.2.3-alpha1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.3-alpha1) to see if the newly introduced MMU Overcurrent Hotfix also applies to your case.

Michele Moramarco
Prusa Research

Prusa-Support, 2025-05-01:
Closing this issue as theoretically fixed by the final firmware release [6.2.3](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.3), pretty much like any other generic MMU Overcurrent error.
Don't hesitate to comment if you think the issue is still valid though.
 
Michele Moramarco
Prusa Research

closed 2025-05-01 by Prusa-Support