prusa3d/Prusa-Firmware-Buddy #4418: [FEATURE REQUEST] Light and filter auto on

bojorno1, 2025-01-19:
### Printer model

MK4, MK4S, MK3.5, MK3.9, MK3.9S

### Describe the feature

The new feature would automatically send the gcode commands for the light and fan using a 2.5mm jack to molex adapter connecting the GPIO hacker board and the Enclosure basic board. There would be a menu option in the printer to enable this functionality if the user would like to control different devices in different ways on Pin 0 and 1.

### Expected functionality

At the beginning of the print, the light and/or fan will turn on, and at the end, they will turn off.

unclej84, 2025-01-30:
I would suggest to have the possibility to start the filter directly once the hot-end is heating up and the loaded filament has the filtration option on in the filament settings. And maybe some overrun (perhaps even as setting in seconds) after the print is finished to clean the air in the enclosure

CZDanol, 2025-01-31:
@unclej84 Please create a dedicated issue, do not hijack other issues. Also, there is already a "post print filtration" option for the XL enclosure.

CZDanol, 2025-01-31:
I've created an additional ticket for the filtration part: BFW-6702. The original BFW-6168 was mostly for the LEDs part. And the BFW-6703 task for the GPIO support.

We are actually at this moment trying to somewhat unify LED controls across our printers - adding support for the GPIO board will hopefully be possible as well.

As for filtration, there will be a filtration fan as an optional accessory for the CORE One as well: https://www.prusa3d.com/product/advanced-filtration-for-prusa-core-one/ and we aim to unify the functionality between the C1 and XL as well. GPIO support should also be possible, hopefully :)