prusa3d/Prusa-Firmware-Buddy #4420: [BUG] XL freezes while in the control panel

Ceethu, 2025-01-20:
### Printer model

XL - multi tool

### Firmware version

6.1.3+7898

### Upgrades and modifications

None

### Printing from...

USB Drive, Prusa Connect

### Describe the bug

Tested on 2-tool and 5-tool Prusa XLs. 
While preparing the nozzle for cleaning, if actions are in order shown below the printers freeze.
Controls > Pick tool
Return >Set temperature
Return > Move axis (here it stops - i can adjust all acces on screen but none of them move, if I choose "return" the printer freezes and I can olny reset it. 

If I switch the order to:
Controls > Pick tool
Return > Move axis
Return >Set temperature - then everyting works just fine

### How to reproduce

Controls > Pick tool > Pick any tool
Return > Temperature > Set adequate nozzle's temperature
Return > Move axis > Try to move any of them

CZDanol, 2025-01-21:
Hey, this doesn't seem to be reproducible on our newest dev firmware, so it probably got fixed somewhere along the way. Maybe it's already working in 6.2? If not, then hopefully the following firmware.

Ceethu, 2025-02-17:
It happened again a few times (on 6.2), on two different XL printers,
always in control panel and always after I heat up the nozzle. The
only "solution" so far is to reset the printer.

CZDanol, 2025-02-18:
Hello, we tried reproducing this on 6.2 and failed. As we are unable to reproduce the bug, we cannot fix it. I'll leave this open in case some new information comes out that could help us locating the bug.

Novaboogie, 2025-04-15:
Bumping this with a scenario that seemingly repeats in locked/frozen control.
XL5T with FW6.2.2

Had a extruder motor stop turning during a print, but no stuck filament error.
I stopped the print and the active tool head parks in front of TH1 (like normal)
In Control menu I manually Moved X to somewhere close to center
Moved Y forward near 0
I set the nozzle temp to 220°C for PLA
Went back to move extruder axis manually and discovered the extruder value was already set to a value north of +1200 without having rotated the knob.
I opened the idler door and could see the extruder gear rotating backwards (unloading).
At this point I could no longer interact with the control, nothing else was selectable. Only option was reset.

A second occurrence happened under the same circumstance (extruder motor stops rotating) and this time I found the extruder value already set to just over +300. Again extruder motor rotating backwards and a locked control. Only option was reset. I should also note that is was a different extruder that stopped rotating.

I'm just scratching the surface in diagnosing, I have not tried to reproduce this without a 'extruder motor stopping mid-print' condition. I could be dealing with a combination of problems. But having a 'frozen' control result from this scenario twice might be useful to the team diagnosing.

CZDanol, 2025-04-15:
How often does this happen to you in the scenario you described?

Novaboogie, 2025-04-15:
Both instances above happened 2 days apart, within 6 hours of printing time, on different print jobs. As soon as I get a chance I'll try recreating this scenario without the 'extruder motor stopping rotation' condition. But at the moment, when I intentionally stop a print, and then try to move an axis, the extruder that parked in front of TH1 will move back incrementally until hard contact/collision is made with TH1, so I'm trying to be cautious.

closed 2025-06-23 by github-actions[bot]