prusa3d/Prusa-Firmware-Buddy #4421: [BUG] XL Multi Tool Does Not Allow Manual Change in Tool 1 Dock Potition

gusbronk, 2025-01-20:
### Printer model

XL Multi Tool

### Firmware version

6.2.0-RC1

### Printing from...

USB [N/A]

### Describe the bug

Tool 1 does not allow manual adjustment of dock position. I understand why T1 does not allow adjustment of nozzle position, but dock position should be adjustable to make minor corrections from the position calibration.
I believe this is a new behavior in 6.2.0 RC1. Error may be on line 454 of toolchanger_utils.cpp, I have not verified this.

### How to reproduce

I updated from firmware 6.1.3. Before updating I reset the entire printer. I suspect the bug would be present despite the update path.

Go to Settings > Tools > Tool 1.
Scroll, notice dock position is not an option.

### Expected behavior

Have an option to manually adjust T1 dock position, same as other tools.

### Files

[image]

CZDanol, 2025-01-21:
Hello, thanks for the report, there is indeed a bug in the logic. Internal ticket: BFW-6641

CZDanol, 2025-01-30:
Hello, I'm sorry, this issue has been mistakenly mentioned in the 6.2.0 release notes. It is fixed, but the fix was only merged to the 6.3 firmware, not 6.2.

gusbronk, 2025-01-30:
Hi, no worries. Thanks for addressing it and all your team’s hard work. Is there a public PR with the fix listed? I’d like to add it to the 6.2.0 release and compile myself. Apologies for having to ask, I’m very new to git.