prusa3d/Prusa-Firmware-Buddy #4422: [QUESTION] about EEPROM, G-Codes, Firmware and z height

asimitabi, 2025-01-21:
### Question

Hello guys,

I have like 30 Prusa Printers and I am happy with them, they are running for years.
I did some changes on two of these printers in the past. One is a Prusa Mini (first generation) and one is a Prusa XL.
I changed the z height a little bit to fit customer projects.
First of all I did it on the Prusa Mini. I did not wanted to flash a custom firmware only for this small z height adjustment, so I did it with G-Code in the Start G-Code. I used the M211 command to ignore the max endstop on the z axis. So far so good, the print continues and can exceed the maximum print height that is in the firmware. Once the printer is finished, of course, the current z height is over the max z height. I can now take the part out of the printer (on the Prusa Mini). When I try to move the z axis with the knob now, it will directly go to the maximum z height that is stored in the firmware. Thats okay as long as you have already taken out the printed part to avoid a crash. That was fine for me because I knew it and could take it out before moving the z axis. That was the case for the Prusa Mini.

Now I did this with the Prusa XL. I added the same M211 Command in the Start G Code. The printer prints the part perfectly above the maximum z height but now with the XL, once the print is finished and the z height is above the maximum z height, it will immideatly go down to the maximum z height (in this case 371,21mm I think. So in this case I dont have time to take the printed part out, I have to be directly next to the printer and stop it to avoid a crash. I tried some thing in the End G-Code to get rid of this behavior but it did not worked. I also tried to store the new z height with the M500 command in the eeporm, the value was not stored and when reading the eeprom the value was still the old one. 

Thats a bit strange as I thought the Mini and the XL are using the buddy. 
My question is: Is it somehow possible to solve this with G-Code?
And if not, I would like to use a custom firmware with the z height adjusted but it seems like that flashing a custom firmware is not yet supported for the Prusa XL. Will it be available in the future? My choice would be to solve it with G-Codes instead of firmware but if thats not possible it would be nice to be able to flash a custom firmware to the printer.

Thank you very much for your help in advance!

CZDanol, 2025-01-21:
Hello, sorry, investigating your issue would take time I currently don't have, but I can at least say that you can flash your XL printer the same way you can flash MK4 or MINI. If you want to use a custom firmware, you need to break the appendix, just as with the other printers.

asimitabi, 2025-01-21:
@CZDanol thank you for your fast reply! Thats good to know as the Prusa Service told me its not possible.

But it is true that its not possible to store values to the eeprom on the buddy board?
Or is there a way, maybe another one than using M500 command.

Thanks and regards!

CZDanol, 2025-01-21:
Our "EEPROM" does not work as the Marlin one. In general, I would say that we are "storing values" to the eeprom, but that's not what you're really asking. M500 does nothing in our FW.

asimitabi, 2025-01-21:
@CZDanol actually, that was exactly what I was looking for. Cant believe I never heard of it or came across this.
Thanks again!

CZDanol, 2025-01-21:
You're welcome :) I am closing the ticket then.

closed 2025-01-21 by CZDanol