prusa3d/Prusa-Firmware-Buddy #4424: [BUG] MMU error screen doesn't trigger "attention" in PrusaLink

JeffFaer, 2025-01-22:
### Printer model

MK4S

### Firmware version

6.1.4+7908

### Upgrades and modifications

MMU3

### Printing from...

SD Card

### Describe the bug

https://github.com/prusa3d/Prusa-Firmware/issues/4470

PrusaLink doesn't signal that the printer needs attention when the printer ends up on the MMU error screen.

### How to reproduce

 - Start a print
 - Trigger an MMU error
 - Notice that Prusa Link still thinks the print is going just fine

### Expected behavior

PrusaLink should indicate that the printer needs my attention

### Files

I'm happy to take a crack at fixing this based on that PR from the other repo, but I haven't yet contributed to firmware like this so I'd definitely like a set of eyes on it to make sure I'm not breaking anything :)

Bigoid, 2025-02-04:
Same behavior with Prusa Connect.... No notification  in the event of failure to change the filament for example. 
It looks like this other issue that was automatically closed due to lack of activity... 

https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4102

nickmayer, 2025-04-07:
I'm seeing the same issue. The API always shows printing even when the MMU requires attention, preventing my Home Assistant integration from correctly notifying me.

CZDanol, 2025-04-07:
Hello, this seems to be resolved by the ticket BFW-5952 which should be in FW 6.2. Is this still happening on the 6.2.x firmware?

JeffFaer, 2025-04-07:
I'm running 6.2.2+8853 and it still happens

nickmayer, 2025-04-07:
I'm running 6.2.3-RC+8861 and am also still seeing it.