prusa3d/Prusa-Firmware-Buddy #4427: [ENHANCEMENT] Speed adjustment of individual feature types

me25542, 2025-01-24:
### Printer model

All

### Describe the enhancement

Currently, during a print the user can adjust the print speed relative to how a part was sliced. This is mostly useful (in my experience) for slowing a print down if it was sliced too fast, making it able to complete. However, usually it is only one thing, like the internal infill, or the support material, that is too fast (or slower than needed). Looking at print files generated by PrusaSlicer, I have noticed that the feature type is indicated before each section of print moves. This is stored as comments in the gcode, but given the comments machine-generated nature, they should be decodable by the printer. I am less familiar with the binary gcode (`.bgcode`) format, but since the gcode I have looked at was converted to ASCII (normal) from binary, presumably the info of what print moves are what type is in there somewhere. It might be in a separate data block for metadata, I don't know. But, given that this info *is* present (at least in files generated by PrusaSlicer), I think it could be exploited.

I have attached an example file (in a Zip folder because it didn't let me attach them directly) that I looked through to show the comments. I do have some custom gcode in there to control a custom enclosure, consisting of some `M260` commands. This can be ignored.

[Shape-Box_0.4n_0.2mm_PLA_MK4S_16m.zip](https://github.com/user-attachments/files/18529863/Shape-Box_0.4n_0.2mm_PLA_MK4S_16m.zip)

### Expected functionality

Users would be able to enable a setting in the "User interface" portion of the settings menu called something like "granular speed control". If this was enabled, and if the file printing was generated by PrusaSlicer (or any other compatible source, perhaps Orca?), then additional speed controls would appear. These might be inside a submenu to avoid crowding the main "Tune" menu. In addition to the global "Print speed" setting, users would be able to change the speed for individual print moves (solid infill, internal (sparse) infill, perimeters, external perimeters, support material, etc.). The global print speed setting would apply to any feature type that the user does not change (and features of miscellaneous/unclear type). individual speed multipliers, if set for any given feature type, would override the global multiplier.

me25542, 2025-03-26:
Still would like this.

me25542, 2025-05-26:
Still want this.

me25542, 2025-07-28:
Still want this.

me25542, 2025-09-29:
This would still be useful.

me25542, 2025-09-30:
You are probably right; the resources on Buddy / xBuddy boards might be too limited for more advanced control over extrusion speeds. It would be nice to see future Prusas with something more like Klipper, or at least more resources somehow. Too bad.

closed 2025-09-30 by me25542

3d-gussner, 2025-10-01:
@me25542 Thanks for understanding and closing this issue.