prusa3d/Prusa-Firmware-Buddy #4430: [BUG]

CloudStrife1983, 2025-01-26:
### Printer model

XL Multi Tool

### Firmware version

6.2.0 RC

### Printing from...

Prusa Connect

### Describe the bug

After a canceled print the Printhead with TH1 was parked in front of the parkspace from TH1. Afer I a new print, the printer make the following weired behavior. **The printhead moves for- and backward but always more backwards than forwards. So he moved further and further back bit by bit It moved backwards bit by bit until it hit the rear parking slot or wall with a loud rattling noise.** There was no crash message and it then drove forwards to its actual homing position and then carried this out normally. 

### How to reproduce

I can't reproduce it because it doesn't happen every time and I don't know what causes it. The error occurs randomly or possibly only under very specific conditions. 

### Expected behavior

It would make more sense to park the toolhead at the front left instead of at the rear left after a manual press abort by the user. Or ask him if he wants to park the TH in the dock.

Graloth, 2025-02-12:
I have encountered this bug as well, but for me I had cancelled it when it was using TH3, it then parked itself in front of TH1.
When starting a new print it OR telling it to park the current TH, it would pretty much ram the active TH into TH1 before then moving back to calibrate it's position for correctly parking TH3.

It causes the locking pins on TH1 to, quite aggressively, get hit against the buddyboard hinged cover.

I've encountered it several times now, both on 6.2.0RC1 and 6.2.0.

If I manage to figure out how to repeatably reproduce it sometime later today or tomorrow, I will post an update with the steps (and test gcode file)

CZDanol, 2025-02-12:
Thank you @Graloth . If you manage to find a good reproducer, that would be awesome.

Graloth, 2025-02-14:
@CZDanol I'm still trying to replicate it, but I think this might be somewhat related to the behavior described in #4259 

I'll still continue trying to get it to happen again, it very much seems like it's random or affected by something that isn't very clear.

Even if the printed parts on my toolheads get damaged by my testing, I've already printed all new parts in PCCF anyway, so I don't mind taking the risk (Since I'm also ready at the power switch anyway, I think it's safe enough)

HEEROCKSS, 2025-03-22:
I also have this problem. 

1. Start print, wait till printer prints. 
2. abort print
3. tool moves infront of TH1
4. Start new print
5. Tool moves a little forward then backwards into TH1

Novaboogie, 2025-03-31:
I also have this problem.

XL 5 Tool
FW 6.2.2
Printing from USB

100% repeatable for me.

1. Stop a print allowing TH2-5 to move in front of TH1
2. Through the control menu, select park current tool
3. TH will move backwards in short increments until hard contact with TH1 is made.

fabianmuehlberger, 2025-05-14:
Very similar but not the same issue on 6.2.4 XL 5T.

1. Aborted a print since it had extrusion issues. 
2. The printhead was in the front left area when it stopped
3. In the menu, I selected unload or purge filament (not sure). 
4. the printhead started moving backward bit by bit for 5 to 10 times. As described by other users.  
5. Error message: HOMING ERROR Y prusa.io/17305. Since there was more space, it did not hit printhead 1 

I assume that the printer looses its position after aborting a print.

Le0Michine, 2025-05-23:
also encountered this, it appears to be related to something printer executes when cancelling a print which went past homing sequence. Probably it parks in from of TH0 and writes current position into memory presumably to optimize homing sequence next time, but instead it rams into TH0 if another tool was attached. If TH0 was selected then it hits rear limit a few times and proceeds to home correctly after that. I wonder what is the cancellation g-code

morphias2004, 2025-07-21:
The XL just isn't getting the attention since the CORE One was released. :(