prusa3d/Prusa-Firmware-Buddy #4431: [BUG] prusa.1o/17536 - Fatal Error on long prints but no persistant issue after reset

MartinMajewski, 2025-01-26:
### Printer model

XL - 5TH - Enclosure

### Firmware version

6.2.0-rc1

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

Hi,

I am encountering an issue with the 6.2.0-rc1 firmware on long prints, where at some point, I get the shown fatal error screen on an random tool head. After a click on the knob, the printer restarts, the print is lost (and all the filament and time), but no issue persists. I can print without any problems new projects, and shorter projects (under 10 hours of print time) finish fine.

[image]

As the print heads are factory assembled and it happens on arbitrary tool heads, I don't think it is the same issue shown here: https://help.prusa3d.com/article/extruder-error-17536-xl_822367

I cannot tell if this issue occurred on the stable FW version, as I haven't had the XL for long enough before upgrading to the RC1 to print long-running jobs.

However, there is a pattern with this issue occurring and long-time prints, which might be a bug in the firmware. As far as I can tell, a crash dump was not created.

Cheers,
Martin

CZDanol, 2025-01-26:
Hello, it is the same issue as in the link you mentioned.
The issue didn't occur at the stable version, because this error has been added in 6.2.0.

Gik-0101, 2025-02-06:
This is the same error as I already wrote.

https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4400

This error has existed for me since FW 6.1.2
But only since FW 6.2.0 RC1 has the error been recognized by the printer and output with an error message.
Prusa has already provided a hardware solution. Please contact support to get the parts you need.
It remains to be seen whether this fix is ​​really the final solution to the extruder problem. I haven't had an extruder failure since the conversion.

I hope this is investigated further. 

Good luck

MartinMajewski, 2025-02-06:
My XL is from December 2024. How recently did Prusa release this fix that it didn't make it into my package? 🤔

Gik-0101, 2025-02-06:
This fix (announcement) was first sent via email on January 22, 2025 to everyone registered with support regarding the extruder failure issue.

So this solution hasn't been around for long. It's still very fresh.

As an alternative, you can also print the parts yourself. Prusa has made the data public on printables.
There are also repair instructions.
You also need a few screws and nuts for the printed parts. But it is listed in the repair instructions. Or you can order the spare parts via support for the affected printer.
But then it takes a while.

https://help.prusa3d.com/guide/how-to-replace-the-main-cable-connector-cover-xl_831370

MartinMajewski, 2025-02-06:
That's awesome! I didn't know about the issue, and when I raised this ticket, the QR code led to a blank page.

I will do the upgrade myself asap!

Thanks, mate!

rovili, 2025-02-18:
> That's awesome! I didn't know about the issue, and when I raised this ticket, the QR code led to a blank page.
> 
> I will do the upgrade myself asap!
> 
> Thanks, mate!

Did the upgrade of the cable connector cover help to resolve the issue for you?

MartinMajewski, 2025-02-19:
> > That's awesome! I didn't know about the issue, and when I raised this ticket, the QR code led to a blank page.
> > I will do the upgrade myself asap!
> > Thanks, mate!
> 
> [@MartinMajewski](https://github.com/MartinMajewski) Did the upgrade of the cable connector cover help to resolve the issue for you?

Some changes in the entire situation:

I was about to do it DIY, but then I changed my mind. All my printers were occupied with print jobs, and I asked myself why I had to deal with the situation alone after purchasing a 4k Euros expensive machine. I reached out to Prusa and asked for support. It was a lengthy chat, and I had to follow the protocol to provide evidence for the issue and fill out a form. I was promised that those parts would be sent to me.

Nearly one month later, nothing. So, I contacted Prusa again and was immediately asked in unbelief why I didn't print those parts myself. That got me mad. 4k on a printer, and you cannot even rely on promises made by the support. If they had told me not to wait for a shipment, I would have dealt with it already, but that got me on a completely wrong path. Therefore, I asked them to send out the parts on principle. This is a warranty case, but they canceled the shipping process internally and didn't even update me.

Therefore, I am currently printing without the original connector cover and haven't had the issue since. However, I also have to admit that none of my print jobs were as long as the ones when the problem arose. I will have to monitor the situation, still. When the parts arrive, I will install them and report the outcome after some time. 🤞

rovili, 2025-02-19:
I did the printed upgrade now, I was able to source the tools via a friend. Although, Prusa should have sent the parts but I couldn't wait.

For me is always E2 and E4 failing, after the upgrading the parts the issue was resolved for E4. Interestingly only during multicolor prints the errors occur and always while the printhead is docked. With singlecolor prints both printheads work as expected 
Which brings me to the question if not the docking is causing the issue?!

rovili, 2025-02-20:
> I did the printed upgrade now, I was able to source the tools via a friend. Although, Prusa should have sent the parts but I couldn't wait.
> 
> For me is always E2 and E4 failing, after upgrading the parts the issue was resolved for E4. Interestingly only during multicolor prints the errors occur and always while the printhead is docked. With singlecolor prints both printheads work as expected Which brings me to the question if not the docking is causing the issue?!

Unfortunately, E4 still fails with the new parts. So E2 and E4 are unusable. The machine doesn’t even have 100 hours on. Got it last week… I am so annoyed.

rovili, 2025-02-22:
The issue started also on E5 now. I am getting on Monday 5 new dwarf cables.

For testing purposes I downgraded from 6.2.1 to 6.1.3 and started a big 3 color print with E2, E4 and E5. There was not a single issue over 17 hours print and 1400 tool changes. No extrusion issues, no errors.

tutur05, 2025-04-27:
Same here, I build new dwarf connector and push latest firmware and the problem is still here... 
Next step, downgraded firmware, not happy but if it's only solution.. :(
I

rovili, 2025-04-27:
> Same here, I build new dwarf connector and push latest firmware and the problem is still here... Next step, downgraded firmware, not happy but if it's only solution.. :( I
Mine is running perfectly fine since 6.2.4.

18pool49, 2025-06-28:
il est vrai que cette alarme ((#17536)) est arrivée suite a une mise a jour ; mais n'a AUCUN rapport avec le logiciel 
la cause est "UNIQUEMENT" mécanique ou physique 
les fiches des câbles des têtes sont "INSUFISAMENT " fixées dans leur socle et occasionne des arrêts d'alimentation des moteurs des "extrudeurs" 
fixation incorrecte aussi sur le châssis pour les "E1 et E2 " qui sont en position horizontale avec le poids des câbles +les vibrations 
les coupures d'alimentation viennent aussi de cet endroit 
pour les têtes j'ai réalisé un clips pour bien verrouiller la fiche dans son socle 

[image]

[image]

en ce qui concerne les sortie châssis E1+E2 je fais usage d'une bride qui assure le maintient efficace dans leur socle les deux fiches 
 
[image]

[image]
[image]

tout cela me donne de très bon résultats  sans aucune interruption suite a plus de 300h de dépose et des milliers de changements d'outils 
je reste a disposition de ceux qui désire les STL de ce clips et bride