prusa3d/Prusa-Firmware-Buddy #4432: [ENHANCEMENT] Enable Spool Join configuration after print has started

TickTock-, 2025-01-26:
### Printer model

XL

### Describe the enhancement

Once a print has started you cannot configure Spool Join.  Jobs sent via PrusaConnect must be cancelled and restarted to get the option to set this up.  Should be able to indicate a Spool Join extruder while the print is still printing.

CZDanol, 2025-01-28:
Internal ticket: BFW-6621

LarGriff, 2025-01-30:
Please add the same for MK4S

CZDanol, 2025-01-31:
@LarGriff of course, if we implement this, it will be for all printers where it's reasonable :)

eqdeng, 2025-02-03:
Adding support. Current Spool Join implementation, and the Print Queue are mutually exclusive.

Naugrimohtar, 2025-02-06:
I second this, I would like spool join to work with the print queue on connect.

Prusa-Support, 2025-06-05:
This seems to be more closely related to an earlier internal ticket: BFW-5273

Lars-del-oss, 2026-11-08:
Still very much needed on **MK4 + MMU3 (FW 8.0.0+17068, Bootloader 2.6.0, MMU 3.0.4)** — and also the ability to **spontaneously switch the active slot during a manual Pause** (`BFW-6621` / #4298).

**Real-world scenario from today:**
- Printing single-material PLA from **Slot 1** (without having enabled SpoolJoin before starting the print), with compatible PLA already loaded and ready in **Slot 3**.
- Near the end of the spool on Slot 1, the filament end was hooked to the spool core and got stuck. Because filament was still sitting inside the SuperFINDA and Nextruder sensor, automatic runout did not trigger.
- I paused the print and unloaded the stuck remainder from Slot 1 via the printer menu.
- At that point, I was completely stuck:
  1. Even though the printer knows both Slot 1 and Slot 3 contain PLA, there is **no way in the Pause or Tune menu to switch the print to Slot 3** or to enable SpoolJoin mid-print.
  2. Worse, after unloading Slot 1 during the pause, pressing **Change Filament** did nothing at all (neither paused nor unpaused), and there was no menu option to make the MMU3 load Slot 3 to the nozzle.
  3. To save the print, I had to manually open the Nextruder idler at the print head, push the filament all the way to the extruder by hand, and close the idler again.

Please allow enabling SpoolJoin and switching the active MMU slot to another compatible slot mid-print / during a pause!