prusa3d/Prusa-Firmware-Buddy #4435: [BUG] incorrect nozzle temperature and layer shift after power outage

tjstansell, 2025-01-27:
### Printer model

MK4

### Firmware version

6.1.3

### Upgrades and modifications

MMU3

### Printing from...

PrusaLink

### Describe the bug

After a power outage on my MK4 using MMU3 (single color, but still using MMU3 profile), my print resumed, but the nozzle temperature was set to 220 instead of 245.  This caused the initial layer to not adhere properly.  Additionally, the print resumed but did not home/align properly as I got a layer shift of about 0.35mm in both the X and Y directions.  Luckily I was next to my printer when the power went out and was able to manually fix the temperature pretty quickly, but the layer shift ruined the 4 hour print job (with 30 minutes remaining).

I'm including the .bgcode that was printing at the time of the power outage.

This looks like the same bug as issue #4096 that never got addressed.

### How to reproduce

I've only experienced the one power outage.  I will try to test some more after my current print is finished to see how repeatable it is.

### Expected behavior

The print job should resume with the correct nozzle temperature and without a layer shift.

### Files

[mk4-power-outage-issue.zip](https://github.com/user-attachments/files/18563664/mk4-power-outage-issue.zip)

tjstansell, 2025-01-27:
[image]
[image]

These show the later shift that occurred.

tjstansell, 2025-01-27:
This could be related to #4310 as well.

bstansell, 2025-01-28:
I just tried a simple print on my MK4 and simulated a power outage.  Temperatures were fine, but the layer shift is noticeable (both X and Y direction)

[image]

closed 2025-04-08 by github-actions[bot]