prusa3d/Prusa-Firmware-Buddy #4436: [BUG] MK3.5 Buddy board cannot be flashed to Mk4 or run 0.9 degree steppers

SimonQuvang, 2025-01-27:
### Printer model

MK3.5

### Firmware version

6.2.0

### Upgrades and modifications

BearExxa v2 (revo micro) 

### Printing from...

Not relevant

### Describe the bug

So first of, this issue started when I decided to add 0.9 degree stepper motors to the Mk3.5. I asked Prusa Support about this as there really was not any information regarding Mk3.5. Support said that it was possible to run Mk3.5 firmware and use 0.9 degree steppers. So I went ahead an bought those. Flash forward 6 months and I get time to install, this is now. I install them and the printer can no longer auto home. This is normally an easy fix, if Prusa hadn't decided to take away almost all modability that Prusa normally brands them self with. 
It is not possible to change microstepping or steps/mm as it is not possible to use M500 to save the values. Even though Prusa list M500 as one of the G-code commands for the buddy board. And even tho the printer happily replays "Ok" and not "Error". 

Okay, what is the difference between my Mk3.5 and the Mk4 I think, well only the extruder. So I can just flash the mk4 firmware, it must be possible to run a mk4 with another extruder. That is very normal thing for people to mod on Prusa's. 

Well the printer throws a fit and says "Error 20611", links to prusa.io/20611 which does not exist. 

```
Firmware/printer versions do not match! You are trying to flash FW meant for other revisions of the Buddy Board. 
Website: prusa.io/20611
```

So I decided to reach out to support yet again today. This time support says that I cannot run a mk3.5 on mk4 firmware (which is weird when Prusa Employees have said that you can run a Core One on Mk4 firmware). However I can just keep using the mk3.5 firmware and run the printer with the 0.9 degee stepper motors. 
When I again reminded support that auto home was not working now after installing 0.9 degree motors I was told some copy paste about it "always" is bearing, belts or pulley. This is of cause not the case. 
I asked how the printer would know to adjust steps and was told the printer would auto detect the 0.9 degree stepper motors. This is not the case. 

Buddy Board version 37 

### How to reproduce

Well, this should be easy unless the issue is my buddy board not working as expected. 

Flash MK4 firmware on a Mk3.5

Install 0.9 degree stepper motors on a Mk3.5

### Expected behavior

I would expect the printer to either auto detect 0.9 degree stepper motors, or having options to adjust microstepping and steps pr mm. 

I would also expect a buddy board (that is the exact same as the mk4) to be able to run mk4 firmware.

CZDanol, 2025-01-28:
Hello, MK4 firmware will simply not work with the MK3.5 extruder. The pinouts are different, the hardware control is different, the probing is different, and so on. Neither will C1 run correctly on the MK4 firmware (and you will be prevented from flashing it).

As for homing working correctly, well unless you configure the printer correctly, it will not work. Currently MK3.5 firmware as it is compiled only considers the OG 200 step motors from MK3. To make it work with the 200 step motors, you would need to change `X_MICROSTEPS` and `Y_MICROSTEPS` to 8 in `Configuration_MK3.5_adv.h` and compile the firmware yourself.

CZDanol, 2025-01-29:
Okay, there should actually be "X/Y axis steps per unit" option in the experimental settings - perhaps give it a try.

SimonQuvang, 2025-02-05:
I had a look at the code and how the firmware works and I can see that there is a lot of logic to the whole printer type and the xbuddy board. A lot more than I had thought. I was initally worried that my board might be damaged since it couldnt run Mk4 firmware (which is should be able to, and which it also can). I really did not think that there was this much logic build for handling installing of firmware. 
That is pretty cool. Good work :) 

There is no X/Y steps per unit option on PRUSA firmware. Only when I compile the firmware in debug mode :) 

I have been told a few times now by support that there should be no problem in running 0.9 Degree stepper motors on MK3.5. It has never come up that I need to make my own custom firmware and thus need to break the appendix (which is not a problem anymore). 
However I used yesterday on trying to change the settings that was needed, flashed the firmware and it somewhat works. The current issue is that it cannot auto home or pass the self test. All the normal things have been checked, so I am fairly sure I need to change some more settings.

Seems like it was caused by some planner function using 400 or 200 step motors to calculate mm_to_rev and the homing feedrate being too high so it crashed to violently into the end stop which made the stepper skip steps. 

So in short you need to change more than just steps per unit and microsteps. 

Would it be possible that options such as "has_400_step_motors" was also added to mk3.5 now when it is needed to be changed? Had to dig really really far to find it.

CZDanol, 2025-02-05:
For the homing issues, I recommend copying stallguard values from the MK4 FW, or something like that.

CZDanol, 2025-02-05:
Internal ticket: BFW-6727. We have thought of a way to hopefully fix this even for the guys who already upgraded.

valentinwittig, 2025-06-09:
Did you find a solution to the problem?
I’m having a hard time adjusting the StallGuard values. In Configuration_MK4_adv.h and Configuration_MK3.5_adv.h, the XY_STALL_SENSITIVITY_MIN and MAX values are identical.

HerkisN7, 2025-08-17:
My solution:
1.  /include/marlin/Configuration_MK3.5_adv.h
#if HAS_TRINAMIC

    constexpr float HOLD_MULTIPLIER[4] = {1, 1, 1, 1};  // Scales down the holding current from run current
    #define INTERPOLATE true // Interpolate X/Y/Z_MICROSTEPS to 256

    #if AXIS_IS_TMC(X)
        #define X_CURRENT 550 // (mA) RMS current.  MK4 motors 
        #define X_MICROSTEPS 8 // 0..256 MK4 motors
......
    #if AXIS_IS_TMC(Y)
        #define Y_CURRENT 700 // (mA) RMS current.  MK4 motors
        #define Y_MICROSTEPS 8 // 0..256 MK4 motors

2. /lib/Marlin/Marlin/src/module/stepper/trinamic.cpp
            if ('X' == st.axis_letter)
    {
        pwmconf.pwm_autoscale = false;
        //LDO 400 step (MK4)
        //370 mA RMS/1phase at 60 mm/min and 5500 mm/min
        //LDO 200 step (MK3.9)
        //258 mA RMS/1phase at 60 mm/min, 332 at 600, 355 at 1000, 372 at 2000, 369 at 3000, 352 at 4000,
        //343 at 5000, 323 at 6000, 314 at 7000, 306 at 8000, 296 at 9000, 250 at 10000, 193 at 11000
        // pwmconf.pwm_grad = motors_400_step ? 0x0C : 0x2E;
        // pwmconf.pwm_ampl = motors_400_step ? 0x12 : 0x1F;
        pwmconf.pwm_grad = 0x0C;
        pwmconf.pwm_ampl = 0x12;
    }
    else if ('Y' == st.axis_letter)
    {
        pwmconf.pwm_autoscale = false;
        //LDO 400 step (MK4)
        //470 mA RMS/1phase at 60 mm/min and 5500 mm/min LDO 400 step
        //LDO 200 step (MK3.9)
        //255 mA RMS/1phase at 60 mm/min, 330mA at 600, 362 at 1000, 400 at 2000, 383 at 3000,
        //386 at 4000, 369 at 5000, 355 at 6000, 345 at 7000, 345 at 8000, 322 at 9000, 250 at 10000
        // pwmconf.pwm_grad = motors_400_step ? 0x0D : 0x30;
        // pwmconf.pwm_ampl = motors_400_step ? 0x16 : 0x1F;
        pwmconf.pwm_grad = 0x0D;
        pwmconf.pwm_ampl = 0x16;
    }

3.  Very important!
Factory Reset > Reset Settings & Calibration

4. Calibration
5.  :)

Herkis

HerkisN7, 2025-08-17:
I forgot:
/lib/Marlin/Marlin/src/module/prusa/homing_cart.cpp, line 335:
// #if PRINTER_IS_PRUSA_MK4()
#if PRINTER_IS_PRUSA_MK4() || PRINTER_IS_PRUSA_MK3_5()
        // If homing is failing, try to recalibrate sensitivity. We do this
        // after we couldn't home perfectly, and increase the perfect only
        // tries so that we still try to home perfectly after recalibrating
        // the sensitivity.

Use Pronterface for monitoring calibration. U can read/write homing sensitivity by M914 (but doesnt rewrite EEPROM data).
I have only StepperOnline 0.9° 0.9A and Usongshine 0.9° 1.3A motors now (LDO on the way). I tested Usongshine motors, all OK. My test printer is made up of leftover parts from converting MK4s to Core One and MK3.5s upgrade and something from Ali. This will be a gift for an elementary school.
When I have more time, I will update my github and add the compiled files.

H.

closed 2026-02-24 by github-actions[bot]