prusa3d/Prusa-Firmware-Buddy #4438: [BUG] Cannot connect to hidden network

HyperTextGo, 2025-01-28:
### Printer model

MK4, XL, probably MK4S

### Firmware version

Latest stable build

### Printing from...

USB, want prusalink/prusa connect

### Describe the bug

I am at a school that has pretty heavy network restrictions (and signin restricions), however the robotics lab has a hidden network that has no restrictions. Our prusas will not seem to connect to the internet. I am not sure if this is a bug or a missing feature. Also as extra info the network name has only numbers. 

### How to reproduce

1. Get a hidden network
2. Manually enter credentials (MK4) or load the config file (XL)
3. Attempt to connect

### Expected behavior

Connect to the internnet

### Files

[wifi config.ini.zip](https://github.com/user-attachments/files/18567196/wifi.config.ini.zip)
If I get time, I'll dump the network config too.

CZDanol, 2025-01-28:
Hello, are you sure that this is due to the wi-fi being hidden? We have some users reporting not being able to connect for temaple to 2PA2/3 mode wi-fis (#3908).

HyperTextGo, 2025-01-29:
The network has a hidden ssid, meaning it doesn't broadcast anything unless it gets the credentials including the name before hand.

CZDanol, 2025-01-29:
I know what a hidden SSID is. I was just asking a question whether it would connect even if the SSID was not hidden.

3d-gussner, 2025-01-29:
- I created to test this a hidden 2.4GHz Wifi network named `Hidden` with Encryption: mixed WPA/WPA2 PSK (CCMP) on my OpenWrt router. Used as test key/password `HiddenPassword`.

- First I connected with my Android phone to this WiFi network and tested one web site on the internet. I had to enter SSID and key manually as it has not have been found by the phone.

- I modified in the `prusa_printer_settings.ini` the `[wifi]` section to

```
[wifi]
ssid=Hidden
psk=HiddenPassword
```
- As soon I inserted the USB drive with the `prusa_printer_settings.ini` on the MK4 got a "wizard" on LCD.
- After following the instructions the MK4 switched to the `Hidden` Wifi network and works correctly.
  - In PrusaConnect under Settings it shows `Wi-Fi SSID address	Hidden`

- Switched back to my normal Wifi
- In LCD -> Settings -> Network -> Wi-Fi -> Wi-Fi Wizard -> Enter credentials manually entered the SSID and key/password manually. Confirmed with Finish to connect to `Hidden` .
  - Also works.

My MK4 has FW 6.2.0-RC1+8787

CZDanol, 2025-01-29:
All right, so the wi-fi being hidden is likely not the cause. Thanks for checking this @3d-gussner

HyperTextGo, 2025-01-31:
I think I may have found the problem, the wifi we have is only 5ghz, and the printer's wifi module only works at 2.4ghz. I'll figure out if I can use a different module, but if that doesn't work, it's not the firmware's fault. I will close this issue with this comment

closed 2025-01-31 by HyperTextGo