prusa3d/Prusa-Firmware-Buddy #4439: [ENHANCEMENT]

HyperTextGo, 2025-01-28:
### Printer model

XL, MK4/S,

### Describe the enhancement

Add a setting that makes the info on the print screen move on faster. This would make it so I wouldn't have to wait around to get the requierd info from the printer. 

[image]

CZDanol, 2025-01-28:
Hello, I would like to get rid of the carousel altogether eventually. I have a task BFW-5403 for it but haven't had free time yet to get into that.

CZDanol, 2025-01-28:
Related issue #4260

HyperTextGo, 2025-01-29:
Ok, if I may ask, can I work on a setting to speed up the carosel until you get around to changing the print screen gui.

CZDanol, 2025-01-29:
You mean that you would implement it yourself? I don't see a problem with that :)

HyperTextGo, 2025-01-30:
the source code is manyly python and cpp right?

CZDanol, 2025-01-30:
The firmware is basically only C++

closed 2025-04-10 by github-actions[bot]