prusa3d/Prusa-Firmware-Buddy #4440: [FEATURE REQUEST] Print speed and extrusion speedometer and speed limiter

murk-sy, 2025-01-28:
### Printer model

All

### Describe the feature

It happens that you very optimistically set some settings for a print, and then not really notice the issue until you're printing something big with a lot of infill. Changing feedrate is an option, but this also slows down outer perimeters (which are relatively slow) and travel, which is usually not needed and may just cause ooze and thus quality issues. Ideally, you just want to limit the max print speed instead of slowing the whole print down.

Additionally, the printer would display the following: (average of last 3 seconds of extrusion moves, updated every 3 seconds)
- Print speed (mm/s) 
- Extrusion speed (mm3/s)
- Type(s) of extrusions

Based on that information (likely calculated live to allow for speed adjustments or perhaps provided as information with every extrusion move), "live" tuning would be much easier and reliable when needed. This information would also in general be more useful to display in the footer during the print than most other available.

Somewhat inspired by this issue:
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4427

Related issues (mid print tuning):
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4275
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3343

Related project (more longer-term alternative approach):
- Adjusting print temperature according to flow https://github.com/sb53systems/G-Code-Flow-Temperature-Controller

### Expected functionality

- Notice part is not printing well/correctly
- Check the current print/extrusion speed
- Adjust extrusion speed limiter (for example from 20 mm3/s to 17 mm3/s)
- Print continues without issues, make a note to correct max extrusion speed in slicer

Tupson444, 2025-02-01:
Very good suggestion. This would be much useful, either if I realize I made bad settings or when testing something.

Another related issue, where I suggested max volumetric speed limit (like in regular Marlin fw): #3636

murk-sy, 2025-02-03:
Extrusion types are currently only marked as comments, but I imagine those aren't processed right now.

I think there's a lot of potential smart features to be explored with it, BFW-5417 being just one of them - maybe something fancy like temperature increase for fast infill printing and whatnot.
Not sure if it's worth processing comments or if long-term just a new gcode should be added?

CZDanol, 2025-02-03:
We are generally discarding comments to optimize the pipeline - so introducing a gcode would probably make more sense, as gcode can be compressed in some nice ways :)

murk-sy, 2026-10-22:
As an addon to this, to perhaps simplify the feature:

In case you're having issues with clogs (perhaps it's a CF material on a 0.4 nozzle), having the ability to reduce print speed but *not* affect travel speeds would be a nice inbetween. This way you reduce potential oozing and reduce the time impact for prints with a lot of objects/perimeters.