prusa3d/Prusa-Firmware-Buddy #4441: [BUG] "Toolchange to tool that is disabled by tool mapping" on 6.2.0RC1

ddavidebor, 2025-01-29:
### Printer model

XL 5 toolhead

### Firmware version

6.2.0-RC1

### Upgrades and modifications

Disabled side filament sensors

### Printing from...

Prusa connect

### Describe the bug

Printing the attached gcode from Prusa connect by going in the file page, then mapping extruders 2,3,4,5 in spool join, I get a error  "Toolchange to tool that is disabled by tool mapping" on 6.2.0RC1

[heated_chamber_PT1_0.6n_0.32mm_PLA_XLIS_16h22m_bed1.zip](https://github.com/user-attachments/files/18588913/heated_chamber_PT1_0.6n_0.32mm_PLA_XLIS_16h22m_bed1.zip)

CZDanol, 2025-01-30:
Hello, at what moment you get that error?

ddavidebor, 2025-02-02:
Hi,

Going from memory, but I believe after the heating. I think it might be triggered because there's some command for extruder 1 in the gcode, but the extruder is not in the tool mapping.

closed 2025-04-14 by github-actions[bot]