prusa3d/Prusa-Firmware-Buddy #4443: XL Extruder Stops Turning  Recovery Method

balloonhotair, 2025-01-30:
### Describe the enhancement

Right now in 6.2 if a XL extruder stops turning the system aborts with a red screen which kills the print.
It would be better if the printer PAUSED with a message telling the user to hit the reset button on the side of the extruder and when the extruders finish rebooting hit the resume function on the console. This way the print can be saved and nothing is lost.

### Expected functionality

Change system abort when extruder stops turning to a Pause with a message.

CZDanol, 2025-01-31:
Hello, options like this were discussed when making the decision and it has been decided to make it a red screen, sorry.

closed 2025-01-31 by CZDanol

balloonhotair, 2025-01-31:
So I guess I like to know the logic of having a 10 hour job fail, when there is a way to save the job and still have the issue be displayed to the user. I know what I proposed works since I used it many times before the version with the error message was created. The problem with the older version I had to watch the machine to see the issue happen and then intervene to correct the problem.  So there was a delay getting it done. Is there any possibility of get a one-off version of the code with my suggestion implemented? I then could test it and report back what I find. I would like to help here. The only other path for me is to fork off a version and download the firmware code and build my own version locally. A lot of work for sure. Please consider my request.

cbryansmit, 2025-02-11:
I bought my 5 head XL for a specific use case.  Many of the parts I intend to print will take about 18 hours.  I just had the first print fail at about 6 hours due to an "extruder not turning error".  It surprises me that it won't just move on to the next extruder in the "spool join" sequence, it surprises me that this is not a recoverable error, and it surprises me the Prusa knows about the error, is aware of a fix, and chooses not to implement a fix.

CZDanol, 2025-02-11:
We have implemented the fix, it is just not a firmware one: https://help.prusa3d.com/article/extruder-error-17536-xl_822367

cbryansmit, 2025-02-11:
Thanks for the quick response.  I'm printing the parts now and am hopeful they fix the issue.

fliedlr, 2025-02-13:
I want to really doubt it is a necessary and userfriendly move to stop with red screen . i have stepped back to version 6.1.3....

balloonhotair, 2025-02-13:
Totally agree the red screen stop is really total BS (Bad Science). They should pause the job and let the user do a reset of the toolhead and resume the job.