prusa3d/Prusa-Firmware-Buddy #4445: [QUESTION] Power Panic when printing from Prusa Connect?

Rebel154, 2025-01-31:
### Question

Does the Power-Panic feature work when printing via Prusa Connect? I have just suffered a brief (15-20 secs max) power cut midway through a couple of prints on my two Mk4S printers (one with MMU3), and neither printer restarted after the power came back on.

I'm aware it never worked when printing via OctoPrint, but I thought it would (should?) work via Connect, as the print file is downloaded to the printer's USB stick.

I was using PrusaSlicer 2.9.0 to send the .bgcode files to the printer via the 'send and print' option (1st in the list). The firmware on both printers is 6.1.3+7898. Printers are now on 6.2.0

me25542, 2025-01-31:
In my experience Power Panic does work when printing via prusa connect, though lately I have notice a slight (~1mm) layer shift on resume.

CloudStrife1983, 2025-02-01:
> In my experience Power Panic does work when printing via prusa connect, though lately I have notice a slight (~1mm) layer shift on resume.

@me25542 Yes, PowerPanic works with Prusa Connect prints. Thats because the printer  always prints from USB. Prusa Connect only copy the printfile to the USB drive. 

The layer shift is due to the fact that the printer no longer has the exact same zero point on X/Y after re-homing. Either it was wrong at the beginning or after the next home.

Rebel154, 2025-03-30:
Just had another brief power cut (20 seconds or so) during a print, and again, the power-panic feature didn't kick in. 

Any advice/suggestions from the developers, please?

Prusa-Support, 2025-06-10:
I suspect it was the case of a "brownout" whereas, unlike total abrupt "blackouts", the power gradually arcs down and therefore, the 
power loss is not detected efficiently, or not soon enough to save recovery data.

If the power interruption is "slow" it wouldn't be ''_leaving enough power in the capacitors to store the position_''.
https://help.prusa3d.com/article/power-panic_2092

Michele Moramarco
Prusa Research

closed 2025-06-10 by Prusa-Support