prusa3d/Prusa-Firmware-Buddy #4447: [FEATURE REQUEST] Custom macros for printer states

me25542, 2025-01-31:
### Printer model

All, though most useful on GPIO compatible

### Describe the feature

On the MK4(s) (and similar), XL, and Core1, there is an LED light bar under the screen to indicate printer status. This is useful for being able to quickly tell, at a glance, the state of the printer. However, especially with printers in an enclosure, these lights can be hard to see.
One possible advantage of the GPIO module sold for xBuddy-based printers is the possibility to control lights, etc. from the printer. A way to have a custom macro gcode file run whenever the printers state changes (printing, attention {filament change}, print done, standby, etc.) would improve the functionality of this product. Users could have an "Attention" light turn on when the printer's filament runs out during a print, or have "Printing" and "Print done" indicators.
In my case, I have a custom enclosure connected via I2C to my MK4S's xBuddy board. It would be great to be able to have the printer communicate to the enclosure when attention is needed.
As explained [here](https://help.prusa3d.com/article/gpio-module_734695#advanced-input-control-and-macro-triggering), the functionality for macros is already somewhat present in the firmware.

### Expected functionality

Similar to how GPIO macros function, users would be able to place gcode files of specific names in a "macros" folder in the root directory of the USB drive.
When the printers state changed (e.g. from standby to printing or printing to attention, etc., basically anything the status LEDs would automatically be triggered by), if the macro file was present on the USB drive, the printer would execute its contents.
Perhaps some commands could be disabled in these macro files, such as printing moves (G0 and G1) and temperature control, as they would need to be runnable during a print, or when the filament was being changed.

CZDanol, 2025-02-03:
Hello, we could theoretically add macro support for print start/stop, but for example the "attention" macro wouldn't be realizable. Attention statuses typically happen during execution of some gcode and block further gcode processing, so injecting other gcodes during that would simply not work.

Regarding print start/stop, you can basically already achieve the same by using the custom start/end gcode in the slicer.

That being said, I have some ideas about possibly being able to "link" GPIO outputs directly to things like this (without going through the gcode)

me25542, 2025-02-03:
Yes, start and stop are already possible with custom gcode from the slicer. The main benefit of macros would be performing certain actions on unpredictable events, such as a filament runout. Directly liking GPIO pins to these events could be a good solution.

anghelos, 2025-04-10:
I'd like to add to this:
I'm currently using the GPIO to turn an enclosure fan on when printing PLA or PETG.
The fan turns on/off with gcode commands on print start and print end.
_However_, if I stop the print manually, the fan stays on, and will stay on until another finished print stops it, or if I start a non-PLA print. Rebooting the printer does not change the state of the fan.

My only solution at the moment is to keep a `turn_fan_off.gcode` file in the USB stick.
It would be great if we could set GPIO commands to be sent either on printer states as mentioned above, or during the boot up sequence.

me25542, 2025-06-16:
Still a feature that would be nice to get in the future, even if it is low priority.

me25542, 2025-10-20:
It appears that this *is* a stale issue. If someone else does not express interest in this feature, I will close this issue as not planed.